arkypita/LaserGRBL #560: Laser is always on.

charliet1010, 2019-01-16:
I am brand new to all this and just got my first laser in yesterday, I believe I got everything correct in GRBL and have $32 enabled but I have no control of the laser or power, it may be a hardware issue but again I have no idea because im pretty new. Everything seems to work just fine other than as soon as I connect the wall power to the laser it is on and wont shut off till i disconnect it, its like TTL isnt doing anything.

einencool, 2019-01-16:
Do you have the same power Unit for laser and the Controller Board?
If you have different units, be sure that they are both connected to ground. Otherwise the laser pwm signal is not on ground and the laser will fire up.

I had the same issue when I began two weeks ago.

charliet1010, 2019-01-17:
So in my case it looks like the power supply for the Control Unit, the one that holds grbl and everything plugs into, does not have a ground on the plug. If that's what your referring to, and that's all it is, that would be amazing!

mintautasb, 2019-01-23:
Same problem here. I will upload pictures of my laser set. There is only one power supply for whole machine. Laser lights on instantly when blue power button is on no matter if it is connected to pc. It seems the controller which is responsible for laser lightning doesnt work. I used 5A main power supply.
[image]
[image]
[image]
[image]
[image]
[image]

mintautasb, 2019-01-23:
One photo more:
[image]

arkypita, 2019-01-23:
If can help: https://github.com/arkypita/LaserGRBL/wiki/Laser-Wiring

mintautasb, 2019-01-23:
Tried to start laser again. Right after connecting it to loptop (power is on, laser lights, fans work full power) I see this. How can I know the reason of this alarm?
[image]

arkypita, 2019-01-23:
Google for "grbl alarm". It'free

brakthehun, 2019-06-11:
The white "signal" pin should not be in the "fan" port.

closed 2019-11-03 by arkypita