arkypita/LaserGRBL #562: Doesn't allow manual COM port entry

vtgit, 2019-01-19:
I'm successfully able to run LaserGRBL in Wine on linux, as well as several other Laser engraver programs originally written for Windows.

However, only LaserGRBL doesn't allow me to specify the COM port manually. It seems unable to detect the proper port automatically -- this is common in the other programs, too. But they allow manual entry of the COM port, and so they have full functionality. 

LaserGRBL is the only one that just shows a blank pull-down for the COM port AND won't let me input the correct one (COM4 in my case). Therefore the program cannot connect to the GRBL engraver. Seems like a very simple change to allow manual entry, and your program could then run on Linux with Wine.

Thank you!

arkypita, 2019-01-23:
Try this version: [LaserGRBL.zip](https://github.com/arkypita/LaserGRBL/files/2787731/LaserGRBL.zip)

arkypita, 2019-02-10:
Please tell us if any success with Wine

closed 2019-11-03 by arkypita