arkypita/LaserGRBL #563: Export gcode

noel841106, 2019-01-19:
Hi, it is possible to export gcode?
I open grayscale picture, set all parameters, and got previev. But i want export all gcode to file instead click start button.
I have 3d printer, and want to put laser next to exteuder. On board i've got marlin, and all is set to accept gcode and engrave using laser. I already done test and working good. But this youre program is the best for me so i want engrave on two machine. I will gcoder put on sd card and connect to 3d printer.
Is this possible in any way?
Thank you.

StuartB4, 2019-01-19:
Yes. Click File menu and Save Program.

noel841106, 2019-01-19:
Wow, so easy! Why I dont find it by myself.
Iam sorry for that silly question and wasting youre time.
Thank you again.

StuartB4, 2019-01-19:
No problem. We all go there. :)

closed 2019-01-23 by arkypita