arkypita/LaserGRBL #565: Edges too dark and fill uniform

Aztaro, 2019-01-21:
Recently tried a very large print with tonal differences in some of the areas, anywhere the laser stopped or slowed was dark but when traveling over any area, even one suppose to be as dark as possible, it all came out one burn density, would switching to an M4 instead of M3 solve both problems or just the edge burning? 
[image]
[image]

PoeticJuicetice, 2019-02-19:
You're doing a horizontal print so everything that ends in the horizontal direction will burn extra due to your accelerate and decelerate rate.
Have you tried altering your PWM in relation to your accel and decel rates?
I have issues with this because I find that burning wood (especially really grainy stuff) does not burn 1:1 in terms of PWM and speed. EX: Half the speed and half the laser intensity will not equal the same burn as full speed and full intensity.

StuartB4, 2019-02-20:
For PWM LaserGRBL generates the GCode "S" power levels between 0 and 255.
Thats what gives you the greyscale.
If you are only using laser power set to 3 out of the 255 then you will have a constant power level.
Prepare your image for burning, then click Save Program on the File menu.
When it's saved open it with Notepad and look at the GCode.
It should have different S values between 0 and 255.

brakthehun, 2019-06-11:
What laser diode are you using?? Diodes 3W+ are very difficult ot get greyscale with.

closed 2019-11-03 by arkypita