arkypita/LaserGRBL #567: Problems with vector trace

WandMaker, 2019-01-22:
Hi
Iv'e found the neatest result for burning larger text, is to do a filled vector trace of a jpg file.  This works well about half the time! the other half, I get an unintended jog, so that the outline does not match the fill.
I assumed this was slippage on the hardware so I slowed the speeds down, but when it happens, it is always in the same place on the graphic despite using different areas of the laser bed.

Anyone have any suggestions what I should try?
[image]

arkypita, 2019-01-22:
Read similar topic: https://github.com/arkypita/LaserGRBL/issues/557

WandMaker, 2019-01-22:
The trace shows a gap between the infill and the outline on the topside of each element. The laser is copying this very accurately, leaving that gap.  Is there any way around this?

[image]

arkypita, 2019-01-23:
@maillemaker if you see the image @WandMaker post on its own first message you can see a picture that show that in different areas of the engraved image he is having different offset in different part of the work-area. The "H" letter is correctly filled while "e, s, t" are engraved differently. This look like very similar to your problems.

> The trace shows a gap between the infill and the outline on the topside of each element. The laser is copying this very accurately, leaving that gap. Is there any way around this?

About the gap shown in preview you must consider that the gap you could see actually could be very very small. I guess you are producing very small output like 2x3cm and you are now showing it very very zommed like this:

[image]

The gap you are seeing in preview are actually tenths of a millimeter, and have no effects on engraving result.

WandMaker, 2019-01-30:
Thanks @maillemaker that is an inspired observation!
I've been having the scenario that test pieces are looking fine, whereas the actual piece is slipping. Obviously the laser doesn't know what it's burning, the programme is the same, so I've been scratching my head, pixies was the best I could come up with!
The test pieces are rougher wood, the work piece is sanded! 
By jove, I think you've cracked it!!

maillemaker, 2019-01-31:
Hope it works out for you, Wandmaker!  I am not convinced yet that my "super-smear" problems are due to work/machine movement, but I am still testing.  So far, so good.

WandMaker, 2019-02-01:
Sorted!
@maillemaker that's fixed it. A bit of blu-tack on the feet is my high-tech solution.

Another small problem I had, was the cables snagging on the bolt-heads and corners, now I'm sure with a 3D printer you could make nice covers, I just used heat-malleable plastic, not so pretty, but works fine!

[image]
[image]

closed 2019-11-03 by arkypita