arkypita/LaserGRBL #568: Generate "Laser Movements" with G1 instead of G0

oxivanisher, 2019-01-30:
On my woodpecker 3.something, GRBL 1.1 was pre installed, which is the current release. But since I use CNCJS on a RaspberryPi, I like to upload the G-Code instead of directly controlling the woodpecker with LaserGRBL. (Just some background information)

The generated LaserGRBL G-Code uses G0 (fast travel) during Laser operation, which is now [disabled by GRBL](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode):

> The laser will only turn on when Grbl is in a G1, G2, or G3 motion mode.

So the generated G-Code simply does not work without manipulating it like this:

From:
```
G90
G0 X0 Y0
M3 S0
F800

```
To:
```
G90
G1 X0 Y0 F800
M3 S0

```
I think this should be included permanently or at least as an option to use `G1` instead of `G0`, since it will be incompatible with all new versions of GRBL.

Also the usage of the new `$32=1` and `$32=1` could be interesting to use. For now, I made a macro in CNCJS for activating and deactivating the Laser mode.

More Information:
* [GRBL Laser Mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode)
* [GRBL Readme for 1.1 changes](https://github.com/gnea/grbl/blob/master/README.md)

StuartB4, 2019-02-10:
> I think this should be included permanently or at least as an option to use `G1` instead of `G0`, since it will be incompatible with all new versions of GRBL.

I'd be surprised if there are any new versions of GRBL, at least for the Arduino UNO version.
The chip that stores GRBL is about maxed out now.

oxivanisher, 2019-02-10:
> I'd be surprised if there are any new versions of GRBL, at least for the Arduino UNO version.
> The chip that stores GRBL is about maxed out now.

This might be, but the current release (1.1, released 2017) is already not working with laser grbl anymore, due to the things I mentioned in my issue. https://github.com/gnea/grbl

arkypita, 2019-02-10:
> The generated LaserGRBL G-Code uses G0 (fast travel) during Laser operation

Wrong: LaserGRBL use G0 only for skip white areas. Any tracing movement are generated using G1,G2,G3.

```
G90
G0 X0 Y0 //positioning
M5 S255
F1000
G0 X77.55 Y4.564 //positioning
M3
G1 X77.672 Y4.573 //engraving
G2 X77.933 Y4.589 I5.656 J-86.045
G2 X78.388 Y4.617 I9.095 J-145.08
G2 X78.881 Y4.646 I10.124 J-170.046
```
Since the first statement is wrong, your whole post is fallacious. 

LaserGRBL is born on v1.1, just to exploit all its best features, and it is 100% compatible with 1.1 version.
About 35% of LaserGRBL user is operating with grbl v1.1 (more than 4K users) and noone is reporting issues, so I think you have to reformulate your report, or better explain what you mean, or understand what's wrong with your specific instance.

arkypita, 2019-02-10:
> Also the usage of the new `$32=1` and `$32=1` could be interesting to use. For now, I made a macro in CNCJS for activating and deactivating the Laser mode.

The equivalent of "macro" in LaserGRBL is "custom button" feature.
You can add a custom button to turn on/off laser mode, and many more.

oxivanisher, 2019-02-10:
Very interesting. I tested it further and realized, this is only on images which will be rasterized. With SVG i get code with G1 in it. But when I rasterize e JPG, Its generating the code in my first post. Am I doing something wrong?

arkypita, 2019-02-10:
I also wanted to try, because a mistake can never be excluded 100%
I have downloaded google logo [from here](http://pngimg.com/uploads/google/google_PNG19645.png) and i have loaded it into LaserGRBL with line2line raster import.

As you can see after the very first lines you can see movement done with G1.

[image]

Then I try with an image, and I found the same issue you are exposing.

[image]

It would seem that the transition to G1 on the very first line is missing. It is strange that so far no one has reported!

arkypita, 2019-02-10:
For what I can see the issue is on the very first line, so not so relevant. In any case I will provide for a fix.

closed 2019-02-10 by arkypita

oxivanisher, 2019-02-10:
> For what I can see the issue is on the very first line, so not so relevant. In any case I will provide for a fix.

Yes, it looked like that for me too.

Thanks for looking into this!

arkypita, 2019-02-10:
Thanks to you for issue reporting!

commit ec3d8f53 (2019-02-10): Missing G1 on very first line

fix #568