arkypita/LaserGRBL #569: Several questions on the interface and G-code

Pj3r, 2019-01-31:
Hello,

I've recently acquired a cheap laser with a 1x1 meter table, and it's been a hoot so far.
On my tiny laptop I have with it, the Eleksmaker software is just a buggy mess, so I went with LaserGRBL, and it runs good, mind you, there's just a couple things I'm missing, most likely due to user error.

Okay, first up (and the most important to me) is the machine zero. How do I set it to x0y0? I guess I can home it, but then the machine zero isn't changed at all.... I don't have any sensors on my rig, so I use a set of calipers and manually move it in the "start" position before turning the motors/program on. I measure 60mm on both x and y spacings, and that's my machine zero. Whenever a cable has snagged, or I have to stop mid-program for whatever reason, and reset, my home gets deleted and it reverts back to machine zero. Quite a bit annoying, so is there G-code or a setting somewhere where I can reset the values?

Second, it would be great if I could have the machine's laser turn on on G1 movement and off on a G0. This would clean up the code significantly, but if it's not possible it's quite alright.

Third, and this is the only feature I really liked with EleksCAM, is it possible to do a "preview" area run? If you don't know what that is, it's basically a square run with the laser on (low power, so S40M3), outlining the area where the laser will be engraving. Basically, it'll go from top left, top right, bottom right, bottom left, top left 3 or 4 times.. so you can make sure the part is in the right location.

Fourth up, I've got a 15w laser, and I do have a spare/different lens (?), but is it possible to engrave metals? I haven't been able to get it to do anything on aluminium/brass/steel yet, but maybe I'm doing something wrong. It might be focal distance, as my machine has no Z axis.

Fifth, does anybody have a clean Fusion360 postprocessor or should I just edit the postprocessor for the Smoothie laser or GRBL? They both seem to add a lot of unnecessary code, and now I end up copying a bunch of G-code between my start and stopping lines. My gut tells me this can be done automagically....

Help on any of these would be highly appreciated (emphasis is on the first tho, that's the main thing)!

StuartB4, 2019-01-31:
> Third, and this is the only feature I really liked with EleksCAM, is it possible to do a "preview" area run? If you don't know what that is, it's basically a square run with the laser on (low power, [...]

Framing, low power Laser.

[image]

Copy and paste this text in to the GCode window.

M4 
S2
F500
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

[image]

You can change the speed and laser power by changing the F for Speed setting and the S for laser power setting, like S10 F200

Tool Tip can be anything you want.
I Use this:
 
Framing - Laser On Low Power. S2. Speed 500.
This will appear when you hovver the mouse over the new button.

Load your image and prepare it for burning, click Connect.
Click you new button and it will move the laser around the outside of your image on low power.

StuartB4, 2019-02-01:
You will struggle to engrave on metal even with a 15 watt laser.
Anodized aluminium should be ok. But most metals will reflect most of the light back off the surface, and the heat produced by the laser is absorbed/dissipated in the metal. So you need a really powerful laser to get good results.
There is a guy on youtube that lasers stainless steel using a 4 watt laser by coating it with mustard first.
https://www.youtube.com/watch?v=B60nXwhK9gk

Pj3r, 2019-02-03:
Thanks man, I'll give it a try first thing tomorrow!
Does LaserGRBL accept the [left] [right], or do I need to enter that data for every program seperately?
For instance, can I give it 4 parameters/popup menu so that i only have to put in the workpiece offset (for instance x100. y100.) and the size of the part from that point (for instance, if the text is 20 high and 100 wide, x100. y20.)?

StuartB4, 2019-02-03:
You dont need to enter different settings for every job.
You just leave the custom button settings as they are.
Load your image and prepare it for burning, including any offsets in the Target image settings.
Then when you use the custom button it will still go around the edge of the image.

[image]

[image]

brakthehun, 2019-06-11:
1. - Machine zero is typically set in the upper right and work is typically done in the negative space.
You can use the $N commands to travel it to the lower left and zero it there,, but it you must be careful that it homes correctly every time.

2. - GRBL1.1 can do that. if you set $32=1,, then only a G1/G2/G3 will activate the laser pin.

3. I really think LaserGRBL should come with perimeter buttons pre-installed. You can follow StuartB4's suggestion.

4. to engrave on metal,, you have to add a laser-able coating to the surface.

5. One of our makerspace members greated a post processor for GRBL with a laser. You can inquire at https://www.facebook.com/groups/snocomakers/

closed 2019-11-03 by arkypita