arkypita/LaserGRBL #570: LaserGRBL line to line quality less than 1

brijeshpanara, 2019-02-01:
I want line every 2 or 3 mm is this possible or possible for generate G-code please support me.

StuartB4, 2019-02-01:
Could you explain a bit better please.

e6910, 2019-02-02:
It would be easier to do that on the image you want to engrave. Then import it using line to line mode.

Marino

brijeshpanara, 2019-02-04:
I want to make a simple dotter using grbl laser insted of laser i use 2-3mm
point bold marker and i want to make image design using dots only when
laser is on my solenoid in place of laser is on and make a dot so i want
gaping or spacing of more than 1mm so i want to reduce lines in mm but this
is not reduced below 0 so this is the problem if you have any solution
regarding this please suggest me.

e6910, 2019-02-04:
a milling machine would be easier. Usiing something like ESTLCAM each dot would be a drill.

brijeshpanara, 2019-02-04:
Is this any software like grbllaser which is input of any image and conver
into dots with specific spacing.