arkypita/LaserGRBL #573: Assembling a Banggood laser engraver with lasergrbl

e6910, 2019-02-03:


e6910, 2019-02-03:
Hi all!

This is not an issue, but I hope this information could be useful for someone. 

I purchased this unit from Banggood: https://www.banggood.com/es/5065cm-Engraving-Area-3000MW-Laser-Engraving-Machine-DIY-Kit-Desktop-Laser-Cutting-p-1268114.html?rmmds=search&cur_warehouse=CN
At US $155.00 I must say it’s a bargain.

Here are my thoughts and learned lessons:

General:
1.	Instructions are something almost nonexisting. There’s a PDF on the page, with plenty of pictures, and few written instructions, but they are almost unreadable. 
2.	There is no documentation on the controller, but it is easy to connect as the steppers are labeled and each connection is identified on the controller
3.	 Parts are not classified. There is no clear part list. I wish they copy Ikea on that.
4.	Structural screws are covered in grease, they must be properly cleaned prior assembly. WD-40 will do just fine.

Assembly:
1.	Sort your components by size.  This especially true with spacers. There are two sizes, almost identical, used on the X axis carriage.
2.	Dry fit everything. Twice. Then again.
3.	Make sure that the plates for the Y axis gantry are mirrored (same side facing forward).
4.	No instructions on how to arrange the cables. IMPORTANT, the laser can burn them if you are not careful. Most people place the controller on the back, but I don’t like that arrangement, as you must reach it to power the power supply. I thinking on setting up some sort of catenary, but I haven’t sorted it out yet.
5.	Frame assembly is close to perfect. Went level and square on first try. Seems very sturdy as well.
6.	Belt tension. It should be taught, but it should allow free movement of the carriages. Pay close attention on Y axis belts, they should have same tension on both belts.

Software:
1.	Tried to give Benbox a try. Couldn’t make it install in English. Decided against it and replaced the Nano with another flashed with GRBL 1.1f.
2.	LaserGRBL worked perfectly, I just needed to calibrate steps per mm. It seems to use microstepping.

Laser:
1.	Powerful
2.	Tiny dot (smallest I’ve seen on a laser diode).
3.	Burns bamboo without much charring. Increase lines per mm to get darker engravings.
4.	LOVED IT!!!

Usage:
1.	Movements are not as fast as set. I haven’t make any measurements, but eyeballing it it seems to be roughly 60%

Overall impressions:
1.	I feel tempted to buy another one… ;-)
2.	VERY satisfied!!

Hope this helps somebody,

Marino

arkypita, 2019-02-05:
PS: be care of your eyes, those glasses does not protect at all.

arkypita, 2019-02-06:
> Which laser do you recommend?

Each laser above 5mW should be considered not safe for eye because blink reflex (0.25 seconds) is not fast enough to make you close the eye before it's too late. This definition is given for the direct vision of the ray, which should not be a thing that could happen so easily.

3b class Laser in range of 5–500 mW is extremely hazardous if the eye is exposed directly, but diffuse reflections such as from paper or other matte surfaces are not harmful, but specular reflections from reflecting surface can be just as dangerous as direct exposures.

A class 4 laser (>0.5W) can burn the skin or cause devastating and permanent eye damage as a result of direct, diffuse or indirect beam viewing. These lasers may ignite combustible materials, and thus may represent a fire risk. These hazards may also apply to indirect or non-specular reflections of the beam, even from apparently matte surfaces – meaning that great care must be taken to control the beam path.

Therefore any laser suitable for engraving could be potentially dangerous for your eye.

http://www.lasersafetyfacts.com/laserclasses.html
https://www.ehs.washington.edu/system/files/resources/lasermanual.pdf

> I need some glasses

Yes you need!

The ones provided by china seller are totally shit. They are not sufficient to offer appropriate protection; on the contrary, they provide false sense of safety.

https://www.youtube.com/watch?v=upCsVrKrW7M
https://www.youtube.com/watch?v=WnDjIDhxnMs

Quality glasses cost at least 50€ and they are marked with CE and-or ANSI/EN signature. In the absence of a high quality glasses, put into practice at least some attentions:

- Do not directly look at your workpiece while engraving
- Focus at the lowest power possible wearing your poor glasses

closed 2019-11-03 by arkypita