arkypita/LaserGRBL #574: Linux support, freelance work

arthurwolf, 2019-02-05:
Hello.

I'm Arthur of the Smoothie project ( smoothieware.org ) and I also sell Open-Source-controller laser cutters ( robotseed.com ).
We've had some users report positive experience with this software ( the main page seem to suggest it's for GRBL only, but Smoothie and GRBL's interface for laser cutting are nearly identical ( Smoothie's has been implemented before GRBL's then copied as far as I know ), user testing seems to indicate LaserGRBL is compatible out of the box with Smoothieware. ).

We'd like to recommend the software to our users ( both Smoothieboard users, and users of our laser cutters in fablabs around the world ), but a few things are missing.
In particular, we'd need the software to be able to run on Linux, as in : having a proper easy way to install it on linux like a .deb.
Also we'd like a few Smoothie-specific features to be implemented, such as upload via network. 

We have the ability to provide free hardware ( Smoothieboards ) to people who'd be ready to improve this situation. We might even be able to pay for work directly depending on conditions and circumpstances.

Would somebody ( lead dev, contributor ) be open to helping with these issues ? You can contact me directly if you want at wolf.arthur@gmail.com, this is open to anyone who feel like they could help.

Cheers !

arkypita, 2019-02-05:
Hi Arthur, I am the inventor and main developer of LaserGRBL project.

The history of LaserGRBL starts when I was decided to build a laser engraver as an hobby project ([here is a picture of my engraver](http://lasergrbl.com/wp-content/uploads/2017/03/DSC_1457.jpg)). All the programs in circulation were either too complicated or too simple, so I decided to write my own software that match all my requirement.

Aware that a program like mine was missing, I thought it would be useful to distribute it to the maker community. Along with the choice to distribute this software I had to make some decisions, first of all to keep the program as simple as possible: few buttons, few options, few configurations, usable without documentation or tutorial.

Since that day LaserGRBL community grow up to more than 10K of user all over the world, confirming that my choices was not wrong.

arthurwolf, 2019-02-05:
Hey @arkypita !

Thanks for the history/context, I was not aware of it, very interresting.

Do you have any thoughts on officially supporting Linux, and potentially officially supporting Smoothieware/adding some Smoothie-specific features ( Smoothie is very popular for laser cutter users ).

Cheers.

arkypita, 2019-02-05:
just gimme the time to write more

arkypita, 2019-02-05:
About your requests:

**Run under linux**
I would be very happy that this goal was achieved. I am aware that many makers use linux, and that the most extreme hardware configuration does not even require the use of a PC, replaced by raspberry & co

LaserGRBL is written in C # code and requires the .NET framework to run in windows environment.
There are two ways to run a .NET executable under Linux: Mono (an implementation of .NET runtime under linux) and Wine (windows emulator that is able to load native executable and .dll, included .net runtime and executable).

I have personally tested **Mono way**, also building the solution with monodevelop under linux environment. I get it working (connecting to grbl and streaming file) but some user interface was buggy because of a bad Mono implementation of .NET TableLayoutPanel that I use heavily in my user interfaces. 
It would require you to redesign the interfaces without using TableLayoutPanel, or find a workaround.
I do not have time to continue this job unfortunately.

I know there are @eltoneo that is trying to get it working under Wine #562 
I give support, but no update from him.

**Smoothie-specific features to be implemented**
As I wrote in the introduction, the primary choice to make LaserGRBL as simple as possible requires a certain "rigidity" in not implementing too many "custom" requests. For example, many users would like more control over the Z axis, custom codes for laser on-off to get it working with pen plotters etc etc

Obviously I'm not completely closed to requests: many of the features of LaserGRBL have been added upon request and user suggestion. Feel free to make a list of the things that would be needed to make LaserGRBL "full compatible" with Smoothie and i will tell you if it is possible (and reasonable) for me to implement them on the "main branch" or if it is better to fork to a "LaserSmoothie" and give that project its own life.

arkypita, 2019-02-05:
P.s. as open-source project anyone can do that if able :-)

arthurwolf, 2019-02-06:
It's very good news that it is .NET, getting software to run on Mono is generally fairly straightforward, Do you know of any way we could make sure this work happens as soon as possible? Including sending free Smoothieboards to you or another contributor to the project, or straight up paying for the work ( we have a limited budget, but we recognize this is quite a bit of work ).

About Smoothie. A few years ago, I wanted to use GRBL to run a lasercutter (this was not a common thing back then ), and contacted the GRBL author what I should do to GRBL to get it to be faster/better for laser cutting/engraving. His answer was to try to port it to 32bit boards. This is how Smoothie was born. 
My point here is that Smoothie runs on more powerful hardware and this has lots of advantages for laser cutting, and in particular for engraving. I really think it would be a really great addition for your software to officially support Smoothie.
As far as I know, the interface to Smoothie is 99% the same as the interface to GRBL. So this is more a matter of "testing to make sure" than a matter of actually coding much.
Would you agree that we send you a free Smoothieboard so you can test your software with it ?

Thanks again for the quick answer, looking forward to cooperating with you, we get some very enthusiastic feedback from our customers about your software, and really hope we can soon get to the point where it becomes one of the main recommended pieces of software for our boards/machines.

arkypita, 2019-02-06:
[image]

This is a picture of LaserGRBL working on a linux machine with mono right now.
This is done on an old version 2.8.27. I should do some test to see how compatible is the actual main version.

This was the problem on UI because of TableLayoutPanel

[image]

arthurwolf, 2019-02-07:
Really cool, it's very good news that it seems there is a single item in the interface that doesn't work. How much work do you estimate it would be to fix ?

You didn't answer about getting a free Smoothieboard. If you want a free board, just email me at wolf.arthur@gmail.com with your address. No strings attached: if you use it for tests or anything else, that's very cool, if you never take it out of the box, I'll be happy also, so no pressure.

arkypita, 2019-02-07:
Even if it is only one element, this is used heavily in all interfaces. TableLayoutPanel is the grid control that I use for the layout of all the other controls, to get the self-sizing and collapsing of UI objects when dynamically hide/show items based on user interaction. In the past I have made long attempts to get it working, but without success. It would require rethinking the interface without using this object.

Today I made an attempt on VM with the latest version of Ubuntu / Mono / Monodevelop and the problem persists. I tried to compile the latest version of LaserGRBL and I see that even the Mono XML support is lack of some classes, so I had to comment on the new SVG import feature from LaserGRBL.
Nothing unsolvable but a lot of work is required for the Linux version, and I'm not sure I want/can do it.

About Windows version and Smoothie support, if you send me a board I will be happy to test it with my code and see how much is needed to do to make LaserGRBL compatible with Smoothie. 
If Smoothie interface is born from grbl I think it will be easy to virtualize some feature to gain the target.

arkypita, 2019-02-07:
N.B: I have browsed a few in Smoothie documentation, but I have not found a simple documentation that explain how a PC program should interface with it.

I am looking for something like this: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

arthurwolf, 2019-02-07:
It should be 99% the same as GRBL. If you find a difference it's very
likely that once reported it would be fixed to be compatible.