arkypita/LaserGRBL #575: append file

Rajmenaria, 2019-02-07:
can we rearrange append file as we desire so we can write text & image easily on single window and check how its look after print
[image]

StuartB4, 2019-02-07:
I did ask Arkypita the developer if it's possible to have a measurement scale down the left side and along the bottom.
That would make it easier to place images/text when using Append file.
As long as you know the sizes and offsets for the files you could place them where you want.

He likes the idea but I know he's always very busy and other adjustments/improvements/fixes/updates etc would probably be done first.
https://github.com/arkypita/LaserGRBL/issues/518

arkypita, 2019-02-10:
@StuartB4 this is not what @Rajmenaria is asking for

> can we rearrange append file as we desire
No it is not possible. I don't know if it will be implemented in the future, it is not a priority. #85

closed 2019-02-10 by arkypita

StuartB4, 2019-02-10:
> @StuartB4 this is not what @Rajmenaria is asking for

Sorry, missunderstood.