arkypita/LaserGRBL #576: Laser doesn't burn (GRBL 1.1)

SocialWood, 2019-02-08:


SocialWood, 2019-02-08:
Hi. New laser (5500mW), and in LaserGBRL everything seems to work properly, but the laser doesn't burn. I did read some reactions here, but I think my problem isn't described (?). The laser is on, but that's it.

SocialWood, 2019-02-08:
[image]

StuartB4, 2019-02-08:
Change $32 to 1.
Make sure the laser is focussed correctly.

SocialWood, 2019-02-08:
Thnx. Focus: how?

Met vriendelijke groet,
Erik Vreekamp

-kort, want mobiel verstuurd-

StuartB4, 2019-02-09:
If you don't have an adjustable lens on the laser the only other way is to raise or lower the laser until you get the smallest dot size possible.
You should have the laser on low power and use eye protection when doing it.

SocialWood, 2019-02-09:
OK, I did adjust it to a small dot. But nothing happens, also not on a paper underground :-(

StuartB4, 2019-02-10:
Can you show us your harwdare and how it is all connected ?

oxivanisher, 2019-02-10:
I might well be the problem I describe in #568 
(Starting with GRBL 1.1, the laser is not turning on with G0 movements, only G1 and higher)

fabbbio986, 2019-02-10:
Maybe a problem with the power supply?
Check if your power supply can give enough current to the laser driver.

SocialWood, 2019-02-12:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

StuartB4, 2019-02-12:
Thanks SocialWood.
Everything looks ok there.
Are you using the power supply that came with the machine ?

Is this the Low Power/High Power Switch for the Laser?
[image]

SocialWood, 2019-02-13:
That's power on/off

SocialWood, 2019-02-15:
Anyone who can help me? It doesn't work in any way :-(

arkypita, 2019-02-15:
> Update: laser is complete down now. I tried to connect to 2-pin laser connection on the Arduino, and after that the laser was off. Demolished?

Laser diode should ALWAYS be powered by its own driver board. In the most ideal form the driver board it is a constant current source, linear, noiseless, and accurate, that delivers exactly the current to the laser diode that it needs to operate. Your driver board is the blue one with fan.

If you have attached the black-yellow cable from laser to anything else you have destroyed it by sure.
Sorry

SocialWood, 2019-02-15:
I connected to the 2-pin laser-connection on the arduino. Is that bad?

But another question: my first problem was not having a powerful laser to engrave in wood and even on paper. How was that possible?

einencool, 2019-02-24:
Hi, 
Was the Code generated by LaserGrbl? What was the „max. Power“ in your Settings for your Picture? In the config I can see, that the „max Spindle speed“ is set to 1000. 
To engrave wood or paper you need almost around 200-250 in your G-Code to engrave wood.

I hope you understand what I mean, but I don‘t know how to explain it :-)

osmosis14, 2019-05-03:
Make sure that when you create the code that you change the 0 to 255 to something more like 0 to 900.

closed 2019-11-03 by arkypita

Actionslats, 2020-06-06:
Hi All. I have the same setup and a similar problem, but my laser is only a 2.5w. It was working, then I updated LaserGrbl, and now it doesn't work. Any ideas?