arkypita/LaserGRBL #577: Suggested enhancement - custom start/end/multiple passes Gcode

fabbbio986, 2019-02-10:
Ciao arkypita,
I’m here to suggest a new possible enhancement: add the capability to add custom gcode blocks like slic3r does.
[Here](https://goo.gl/images/M43dMm) it is what I mean.
I was thinking that could be interesting to add 3 customizable blocks:
- start gcode: gcode that would be performed only one time at the beginning of the program;
- end gcode: gcode that would be performed only one time at the end of the program;
- multiple passes gcode: gcode that would be performed only between two passes.

I think that in this way some requests could be merged,like for example: [#348](
https://github.com/arkypita/LaserGRBL/issues/348) and [#95](
https://github.com/arkypita/LaserGRBL/issues/95)

I hope this helps!

Ciao!

arkypita, 2019-02-11:
like!

arkypita, 2019-11-23:
Added. You can test this version: https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.19

closed 2020-01-15 by arkypita