arkypita/LaserGRBL #579: Missing Configurations/cant enable laser mode

monk1974, 2019-02-13:
Hello everyone. Quick summary, got a diy laser weeks ago, default software wasn't good, changed control board, started using LaserGRBL <-loving it.

My problem is I have been using this forum to figure things out and I have realized that everyone's screenshots of their grbl configuration window doesn't look like mine.

I am missing some values and this has made certain things impossible to do. I did not delete anything, this is how its been since install. I think my laser is out of laser mode and cant change it.

[image]

arkypita, 2019-02-13:
Laser mode was added in grbl v1.1.
What grbl version is installed in your control board?

https://www.google.it/search?q=update+grbl+to+1.1

monk1974, 2019-02-13:
My card says Woodpecker 3.0.2  GRB
Apparently it is a 0.9

Thanks for the reply

Gabezoid, 2019-03-02:
How can I check the installed version?
I have problem with jogging command, but maybe I just need to update the board

StuartB4, 2019-03-02:
[image]

Gabezoid, 2019-03-02:
Oh damn me, it was easy! :) Thank you very much Stuart, I'll open a thread then for the jogging problem then, my board should be updated (i have 1.1z version)
Thank again!

StuartB4, 2019-03-03:
I think 1.1z is a custom version, I would stick to official 1.1f from here:
https://github.com/gnea/grbl/releases

Gabezoid, 2019-03-03:
Indeed the problem was the 1.1z! Thank you very much :)

closed 2019-03-08 by arkypita