arkypita/LaserGRBL #581: Override interface

oliv690, 2019-02-17:
Hello,
I tried to use override inetrface but it doesn't appear, even if I have LAserGrbl V3.0.4 and Grbl 1.1g.

I checked config.he and I don't disable any information on override setting.

Could you help in on this issue?

Than you

StuartB4, 2019-02-17:
[image]

LazerGalva, 2019-04-25:
Hi, I have something similar, but I don't have a override interface even on the bottom. What I should do now :D ? Is it some ERROR, or i need some plugin or fix bug? Actually I need all those buttons as well, dose someone knows some Gcode for that?  Thank you!
[image]
ug.

closed 2019-11-03 by arkypita

cherry1pita, 2021-12-15:
a little late to this conversation. just started using lasergrbl and wanted to ask what the linear and the rapid options were for .?