arkypita/LaserGRBL #582: Laser engraves unevenly

StepanTsymbaliuk, 2019-02-19:
Hello!
Recently during the engraving with LaserGRBL I have some problems with the final result. The thing is that in the middle of the process the engraving letters are shifted. The programmed picture is dislocated randomly and some extra lines are engraved. Please see the attached pictures.
[image]
[image]

arkypita, 2019-02-19:
Not a software problem, check your hardware.
Belt, motor, vibration and slippage of engraver respect of the piece or vice versa...

https://github.com/arkypita/LaserGRBL/issues/557

StepanTsymbaliuk, 2019-02-20:
Thank you for your quick response. I appreciate your help. Could you also advise what is the reason why some extra lines are engraved along with the main letters?
[image]
[image]

StuartB4, 2019-02-20:
Looks like the laser is not switching off when moving across a blank space.
Was the GCode generated with LaserGRBL of another application and loaded in to LaserGRBL.

StepanTsymbaliuk, 2019-02-21:
No, I never use GCode or any other application generated with LaserGRBL. Will Gcode help me to solve this problem? Are there any other applications that can improve the engraving?

StuartB4, 2019-02-21:
You could try a program called T2 Laser.
They have a free trial version, so you dont have to buy it to try it.
https://t2laser.wordpress.com

SHristov92, 2019-08-04:
The misalignment looks like missed steps. Check the current settings of the motors and if there is some mechanical issue that could block the motion.

StepanTsymbaliuk, 2019-09-27:
Hello!  Thanks for all the help, but the problem that the laser is not switching off when moving across a blank space is still present.  I was told this could be when the Adruino settings on a computer is TTL and on laser heads it' PWM and they can't work together.  It is possible?  How can I fix this? 
Thank you for your help in advance. 
[image]
[image]
[image]
[image]

arkypita, 2019-09-27:
> Hello! Thanks for all the help, but the problem that the laser is not switching off when moving across a blank space is still present. I was told this could be when the Adruino settings on a computer is TTL and on laser heads it' PWM and they can't work together. It is possible? How can I fix this?
Thank you for your help in advance.

TTL or PWM mean the same. They are two words that say that the laser is able to be switched on/off rapidly via 0-5V signal ([TTL](https://en.wikipedia.org/wiki/Transistor–transistor_logic)) so you can modulate its power by switching on/off very fast and varying duty cycle ([PWM](https://youtu.be/B_Ysdv1xRbA?t=10)).

Laser modules with this ability usually have 3 Wires (black/red for power + other color for TTL signal).

In the image you show seem that the laser does not turn off "randomly", not always.
What append if you run the same program twice? the laser does not turn off always in the same position, or not? In few words: can you confirm that this is a random effect or not?

Did you see any red error icon near sent GCode?

[image]

StepanTsymbaliuk, 2019-09-27:
@arkypita The laser makes a random error.  There are no constant errors.
 No.  unfortunately i didn't pay attention to the error icon (if it fails again i will definitely check)
 if you run the program for a second time (after an error) it is possible that everything will be fine or again it will make a mistake somewhere else.
[image]
[image]
[image]

arkypita, 2019-09-27:
> There are no constant errors.
> No. unfortunately i didn't pay attention to the error icon (if it fails again i will definitely check)
> if you run the program for a second time (after an error) it is possible that everything will be fine or again it will make a mistake somewhere else.

LaserGRBL always send the same commands.
Also your electronics do always the same stuff for the same commands.

So What?

When LaserGRBL send a command to your electronic sometime could happen that the command are received wrong.

This is like what append in the "[chinese whisper](https://en.wikipedia.org/wiki/Chinese_whispers)" game: You say "cat" but the other ear "rat".

If both words have a common sense in the phrase you don't have any chance to realize the mistake.
- I turned the corner and I saw a really big cat
- I turned the corner and I saw a really big rat

The hardware give no feedback of what receive, except for a simple "ok" / "error" message, that LaserGRBL report as green or red icon. 

So, if sometime you have a red icon near any message, this could be the issue.

StepanTsymbaliuk, 2019-09-27:
@arkypita I understand what you're trying to say. But the thing is that I don't know how can I avoid this error happening again? By error I mean Laser is not switching off during the engraving and making these extra lines.

arkypita, 2019-09-27:
> the thing is that I don't know how can I avoid this error happening again?

If the problem is what I assume, you can try the solution explained in lasergrbl.com/faq

StepanTsymbaliuk, 2019-09-27:
@arkypita Thanks for the advice. I'll try to solve the problem. I'll keep you in touch.

closed 2019-11-03 by arkypita