arkypita/LaserGRBL #584: Over Travel to compensate for Darker edges

PoeticJuicetice, 2019-02-19:
First off I want to thank the LaserGRBL team for all your hard work on such a great piece of software. Much appreciation for what you have created and continue to support.

I know the issue I'm about to describe is something I could be adding to my GCODE myself (and I do sometimes, but have yet to build a decent macro for).
I see a lot of issues on my work and in others in these threads where the edges are burnt more than the rest of our images due to acceleration and deceleration of our stepper motors.
Adjusting PWM as the velocity adjusts just isn't fixing the issue, either - thresholds, burn rates, etc. (especially on really grainy woods) 

I was wondering if the user could get the option to over travel at an intensity of 0.
For example if we were to burn at a horizontal direction and it normally ended at X100 S120 we could have the option to over travel a certain distance so now the new GCODE sends the X100 S120 then X101 S0. 
Obviously it would then shift in the Y direction and do the same for accelerating - X100 S0 then start normally X? S120

Each user's stepper motor's accel and decel rates are different so over travel distance input may be an option as well.

arkypita, 2019-02-19:
Acceleration offset/border feature is already in dev roadmap

arkypita, 2019-02-19:
https://github.com/arkypita/LaserGRBL/issues/64

PoeticJuicetice, 2019-02-19:
Thanks arkypita, didnt know if Acceleration Offset or the Power on Delay could be referring to this issue.
Keep up the great work

closed 2019-02-20 by PoeticJuicetice