arkypita/LaserGRBL #586: Help to start the laser. , SW-LD73 v1.1, with DEVELOPOWER V2.0

josestrella, 2019-02-22:
Good afternoon, I bought this laser module and I have this arduino card. I am new in this world and I would like to know how can I put it to work?
[image]
[image]

Cetarik, 2019-10-13:
First, let my say that i am not 100% sure but i can tell you that i have just made my device work with DEVELOPOWER v2.0 but different.
My board looks like this:
[image]

As you can see they are quite the same.
I guess you need to desolder that thing in the laser slot, solder there 3pcs of 2,54mm pins.
Then you need to measure what is your pinout. Middle pin should be GND but for example my board has switched PWM and +12V as you can see in the picture.
For me now it works with my board.

[image]

closed 2019-11-03 by arkypita

jeffolemes, 2019-11-17:
Oi amigos, sou novo no assunto e gostaria de uma ajuda se possível, tenho o mesmo módulo laser, um arduíno uno e CNC shield v3, o grbl está configurado para laser já confirmei tudo mas quando ligo na energia já saí queimando na potência máxima, não faz o processo de liga e desliga na gravação com os comandos M3, M4 e M5! Por favor

leoheck, 2019-12-21:
Hi @josestrella I have the same board. I am looking for the schematics of this board. Do you have any idea how to find it? I would like to work around it to be able to use GRBL 1.1

johnalleO, 2020-02-13:
I purchased one of these and am looking at https://madhacker.org/sw-ld73-laser-driver-modification/ where they talk about removing a resistor to stop full power on start up. Has anyone tried this and does it work? I want to make sure that if I do remove this that the power will get to 100% when I need it.

StuartB4, 2020-02-14:
Not tried it myself, but you should know they remove a Diode the Resistor is next to it on the right. If you do remove it and end up putting it back on make sure you put it the right way round, take a photo before you start. Diodes are polarised so they will only allow current in one directiuon, so it has to be right.