arkypita/LaserGRBL #588: First connection

Rey8801, 2019-02-26:
Hi,
I got a small 2.5w laser before decide to go to a 15w. It's the classical from banggod with the 2 axis frame and Arduino nano commanded. The seller suggested to use benbox as software and indeed it does work, although I do not really like it. It is unfinished and most of the functions are not supported. I have see this open source program and I think it's great. Only issue is that I can not connect it. When I try I find the right COM port and the baud rate set it to 115200. Then press connect the laser binks once and try to connect but then doesn't succeed. I am sure it is something simple since a lot of people use the same laser l, but I can not understand what. Drivers are installed. 

Thank yup for the help. 
Best, 
Alessio

arkypita, 2019-02-26:
Ciao alessio, confido che tu abbia già letto le FAQ sul sito lasergrbl.com

Rey8801, 2019-02-26:
CIao,
NO Infatti sono un ciula, non avevo caricato la libreria grbl sull'Arduino nano. Ho dato per scontato fosse gia' installata. Ora si connette tutto ok. Solo una domanda come faccio a farli fare un disegno con la stessa intensita' del laser. Per capirci se carico un bpm che visivamente non ha totalita' di grigio, e' semplicemtne nero, quando genera il gcode vedo S160/S187/S255 ecc.... Come faccio a farglielo generare tutto a S255 (max potenza). Il laser che ho e' talmente poco potente che altrimenti non mi segna nulla. Grazie mille e bellissimo programma!

arkypita, 2019-02-26:
Ciao, almeno te la sei cavata da solo a trasformarla in grbl.
Ho fatto uno strappo alla regola, ma qui si scrive in inglese. Inoltre, meglio creare diversi thread per diverse domande, così da essere utili a chi li cerca in futuro.

[image]

Even if an image looks black, it is often not. If you zoom in, you may notice that anti-aliasing on the edges produces shades of gray. 

In addition, even if your image was totally black, the scaling performed by LaserGRBL to adapt the image to the required dimensions would produce shades of gray on borders.

This is perfectly ok, and should not afraid you. Those S160/S187 etc only act on very small segments, and helps to do a smoother border effect, instead of grain and scaled ones.

However, if this effect is not appreciated, there are some way to act on it.
The first and the one I suggest is to overcharge S Max setting in raster import procedure (see S300).

[image]

This will produce higher S values that grbl accept as "max power" without any problem.

The second one is to use power override 

[image]

Rey8801, 2019-02-26:
OK perfect, thank you very much. No problem for me. I started the thread in English but you replied in Italian, LoL. That's why I switched :-). Anyhow I will try. I guess my laser is just pretty weak. Once I will be familiar with it I will probably replace it with a decent one.

Rey8801, 2019-02-26:
Sott I just wanted to say that I can close the issue now. Thank you for the support.

arkypita, 2019-02-26:
you are welcome

closed 2019-02-26 by arkypita