arkypita/LaserGRBL #589: Set the correct image size

Rey8801, 2019-02-27:
Hi, just started with laser engraving and the program is great. Only thing the step/mm set are not correct and the works come out with a different dimension than the one decided. For instance the one attached should be 20mm on the long side and it came out above 5. I am sure it's just setting but I looked and I ma not sure where to change it and I got the laser has second hand and the guy does not remember which was the the link to the product so I do not know the step revolution from the seller. It looks like this one https://www.aliexpress.com/item/50-65cm-Mini-3000MW-Blue-CNC-Laser-Engraving-Machine-2Axis-DC-12V-DIY-Engraver-Desktop-Wood/32847761533.html?spm=2114.search0104.3.9.44a11e7b142tgH&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_319_317_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10902_10843_10059_10884_10887_321_322_10103,searchweb201603_16,ppcSwitch_0&algo_expid=c36c24b8-b9f0-4c6a-8261-c5a6cc0991c0-1&algo_pvid=c36c24b8-b9f0-4c6a-8261-c5a6cc0991c0&transAbTest=ae803_4
sold on Aliexpress or banggod with Nema17 motors, but I do not know the motor specs.

Thank you for the help.
Best,
Alessio

Rey8801, 2019-02-27:
For some reason I can not attached a pic.

StuartB4, 2019-02-27:
This spreadsheet document is a stepper calculator and is really accurate.
https://my.pcloud.com/publink/show?code=XZdldd7ZfaIhSLfgCauotkKlC347uYxRxFBX
You can load it in to Microsoft Excel but it also works with Libre Office which is a free office program.

Here's how to use it.

[image]

To enter the new settings in GRBL you can do it from the GRBL menu in LaserGRBL.

[image]

[image]

Rey8801, 2019-02-27:
Man that's fantastic! Thank you a lot.

Rey8801, 2019-03-02:
Thank you Stuart, the excel sheet worked perfectly. Now the laser is really precise. What I notice It does really well with vector toolpath but when I try to laser engrave pictures using orizzontal line it doesn't work really well, tends to do not burn along the longer horizontal line. it's probably a setting problem that I have to find the sweet spot but so far either I burn the edge too much to get the central line or I do not get it at whole.

StuartB4, 2019-03-02:
Do you have Laser mode enabled in GRBL settings.
If not set it here:

[image]

[image]

[image]

Rey8801, 2019-03-03:
Hi thbaks for the info. Always really clear. I will check tomorrow. I think the laser is set to gbrl since I can see the changing in laser power. One thing I know is thst I was using M3 and not M4. Not sure I have the M4 option available but I will check. Which is the min differ ence between M3 and M4? Thank you!

Rey8801, 2019-03-03:
Hi Stuart, thank you. It works now!

StuartB4, 2019-03-03:
Hi Rey8801.
That's good to know. Glad to help. 

Info on M3 and M4

> 

M3 Constant Laser Power Mode:
Constant laser power mode simply keeps the laser power as programmed, regardless if the machine is moving, accelerating, or stopped. This provides better control of the laser state. With a good G-code program, this can lead to more consistent cuts in more difficult materials.

 For a clean cut and prevent scorching with M3 constant power mode, it's a good idea to add lead-in and lead-out motions around the line you want to cut to give some space for the machine to accelerate and decelerate.
NOTE: M3 can be used to keep the laser on for focusing.

M4 Dynamic Laser Power Mode:
Dynamic laser power mode will automatically adjust laser power based on the current speed relative to the programmed rate. It essentially ensures the amount of laser energy along a cut is consistent even though the machine may be stopped or actively accelerating. This is very useful for clean, precise engraving and cutting on simple materials across a large range of G-code generation methods by CAM programs. It will generally run faster and may be all you need to use.

Grbl calculates laser power based on the assumption that laser power is linear with speed and the material. Often, this is not the case. Lasers can cut differently at varying power levels and some materials may not cut well at a particular speed and/power. In short, this means that dynamic power mode may not work for all situations. Always do a test piece prior to using this with a new material or machine.

When not in motion, M4 dynamic mode turns off the laser. It only turns on when the machine moves. This generally makes the laser safer to operate, because, unlike M3, it will never burn a hole through your table, if you stop and forget to turn M3 off in time.

Rey8801, 2019-03-03:
Perfect! All clear. I wanted to show you the result I got but I can not upload pics. Probably I just do not how to do it on github. Thank you very much for your help

StuartB4, 2019-03-03:
[image]

Rey8801, 2019-03-03:
[image]
ok now it worked. When I tried last time didn't. The top left is with your suggestion. I think I only need to go faster to get more resolution. For the rest looks good to me.

StuartB4, 2019-03-03:
Yeah looks good. I’m glad you are getting on ok with it. We all have to learn when it’s something new to us. I’m still learning myself. Good luck with it.

Rey8801, 2019-03-03:
Absolutely, learning is part of the process. I will then close the issue since you answered to all my questions.
Thank you again, see you at the next issue :-)

closed 2019-03-03 by Rey8801