arkypita/LaserGRBL #590: ESP 8266 to CNC shield on top of Arduino Uno

JalenTk, 2019-02-27:
I'm planning to add Esp 8266 to a CNC shield on top of an Arduino Uno. I wanted to use the laserweb ESP8266 as shown in the guide. Is there an schematic diagram on how can I connect my ESP 8266 to the CNC shield? 

Another question is should i upload the websocketserver.ino to the Arduino Uno attached with CNC shield along with ESP 8266?

Thank you in advanced! #

arkypita, 2019-02-27:
> Is there an schematic diagram on how can I connect my ESP 8266 to the CNC shield?

Arduino TX to ESP RX and arduino RX to ESP TX, with a voltage divider for level shifting, exactly as in the picture you can find in here: https://github.com/arkypita/ESP8266-SerialWebSocket

ESP should be powered at 3.3v so you can power it using 3.3v regulated out from arduino board. Just remember to place a capacitor (100uF 5v could be great enough) near the ESP vcc.

> Another question is should i upload the websocketserver.ino to the Arduino Uno attached with CNC shield along with ESP 8266?

To upload websocketserver.ino to the ESP (not to arduino!!!!) you should follow any of the hundreds of guide you can find googling for "how to program ESP8266".

JalenTk, 2019-02-28:
> 
To upload websocketserver.ino to the ESP (not to arduino!!!!) you should follow any of the hundreds of guide you can find googling for "how to program ESP8266".

Just to make things clear sir, so I'll have to upload websocketserver.ino to the ESP seperately (http://lasergrbl.com/en/usage/wifi-with-esp8266/) before connecting it as shown (https://github.com/arkypita/ESP8266-SerialWebSocket) to Arduino Uno right?

pblacq, 2019-04-11:
If you are using an ESP12 module that has an integrated USB<->RS232 converter on it, you need to disconnect the RX signal between the ESP12 and the converter, when you connect to the Arduino.

You need to program the ESP12 (with websocket) before you wire it to the Arduino.

closed 2019-11-03 by arkypita