arkypita/LaserGRBL #591: Custom scripts / macros

brakthehun, 2019-02-27:
So,, where you choose M3 / M4 for laser on and M5 for laser off,, it would be nice if there was another selection called "Custom" so people with dual lasers can use their coolant / mist pins to run their 2nd laser. My idea would be,, you go to the settings pane and there's a place you can type in the two settings. The "laser on" command,, whether it be an M6 / M7 or they can type in G0Z-10 for their Zaxis pen plotters. [power] would be added.

arkypita, 2019-04-02:
Hi @brakthehun 
Custom codes for laser on / off is already on my mind and in devel roadmap #64 
However, implementing it is not so simple as it sound.

What codes is used to turn on/off laser is widely used inside the program to drawing image preview, restore state after job resume and many more (i don't remember now, but i have done an analysis some time ago and there are a lot of part involved).

arkypita, 2019-04-02:
However it is already in roadmap, so I close the issue.

closed 2019-04-02 by arkypita