arkypita/LaserGRBL #596: Jogging issue / Doesn't move

Gabezoid, 2019-03-02:
Greetings everyone!
I'm testing laserGrlbl on my Bachin Maker, and looks great!
Just one thing, jogging control doesn't respond.

The machine is connected and as you can see on the image it recognise the commands., but when I press it the jog doesn't move. What can I do?

Thanks everyone for your time :)

[image]

StuartB4, 2019-03-03:
Although it works with my machine with and without the spaces, have you tried typing the commands with spaces in like this.
$J=G91 X10 Y20 F200

Gabezoid, 2019-03-03:
Solved! I updated to 1.1f and it works!

StuartB4, 2019-03-03:
Thats good to know. :)

closed 2019-04-02 by arkypita