arkypita/LaserGRBL #597: Differences between vectorise and centerline gcode

Haffa4, 2019-03-04:
Version v3.0.9
Gcode saved to sd card and used in a genuine Prusa Mk3 printer.

When I use vectorise and centerline to produce gcode for the same image they both look ok in preview but the saved gcode is quite different.

Vectorise produces this as an example.
`M3
G1 X96.45 Y145.701
G1 X78.05 Y145.675
G1 X59.65 Y145.65
G1 X59.625 Y134.775
G1 X59.599 Y123.9
G1 X78.025 Y123.9
G1 X96.45 Y123.9
G1 X96.45 Y134.8
M5`

Where Centerline produces this
`M3 S255 
G1X21.3F150 
X21.4
X21.5
X21.6
X21.7
X21.8
X21.9
X22
X22.1
X22.2
X22.3
X22.4
X22.5
X22.6`
**These two snippets are not equal, they are just examples.

Vectorise puts the G1 command on each line and includes both x and y on each line.
Centerline puts a G1 only on the first working line and does not always include both x and y.

While both formats may be valid gcode the result for me is that my 3D printer can run the vectorise gcode fine but cannot run the centerline gcode. Is there a way to make centerline produce the same gcode format as vectorise?

I am a first time poster here, forgive me if I miss any required information, please ask and explain if you need more. Thank you.

Haffa4, 2019-03-13:
Hi, Any thoughts on this yet? I'll need to write something to transform the gcode it does create if you do not think it can or should be done in LaserGrbl. Cheers, Mark.

arkypita, 2019-04-02:
Hy @Haffa4 
Both centerline and vectorise use huge parts of code not write by myself (and not so well written in my opinion). I don't exclude to have a look at the issue you signal but when i have time.