arkypita/LaserGRBL #598: Lazer software won't connect

haunna, 2019-03-05:
The laser GRBL software, while it says it's not connected still causes the Z axis to bottom out as soon as I click the connect icon.
I can't stop it and it will continue to run, trying to go lower even after it bottoms out. I have to unplug the machine.
I also have to unplug the USB or when the machine is re-plugged in it, starts to try and bottom out again.
So it would seem there is some connection, but an improper one.
Any Ideas. 
Everything is new, Router CNC GRBL works greats, Just this problem with the laser.
Ernie

StuartB4, 2019-03-05:
Have you got Hard Reset [DTR+RTS] enabled in LaserGRBL settings.
If so try it with it disabled.

[image]

haunna, 2019-03-05:
Stuart,
I got the program to run whatever pic I choose doing as you suggested. I click 'on" the hard reset.
Everything loads and stays connected; But the laser doesn't come on.
I have control of the manual controls and it will run the file, just No Laser.

Ideas ??
Ernie

StuartB4, 2019-03-06:
What connections do you have to the laser.
Can you upload pics of your setup, like the wiring connections for the laser / laser driver and the main control board.

closed 2019-11-03 by arkypita

codechangerr, 2022-08-20:
did you turn on lazer mode with $32=1?