arkypita/LaserGRBL #599: CNC Router added to Laser grbl

Rick23n, 2019-03-05:
I have replaced my 15 watt laser with a 500w 48vdc ER11 Brushless spindle motor. Can anyone help me to control this using laser grbl software? The 'X' and 'Y' axis control should be ok but how do I bring the 'Z' axis into the"G" code ? I want to be able to engrave 5mm acrylic to a depth of 1 to 2mm
Any help would be appreciated
Cheers Rick

Rick23n, 2019-03-06:
Any suggestions as to why my laser motors are locked when in the "idle" mode and the "X" axis stepper is running red hot to touch?

StuartB4, 2019-03-06:
Hi Rick.
You may need to reduce the current to the X axis motor.
This video will show you how.
https://www.youtube.com/watch?v=89BHS9hfSUk

Rick23n, 2019-03-10:
Thanks Stuart
It automatically unlocks when the 5v is applied via the USB. I still have not found out what program I can run with grbl to operate my router. As I said previously, its the "z" axis control that I need to engage and lift the router via g code but I haven't been able to find anything that will work with the software I have.

StuartB4, 2019-03-10:
>  I still have not found out what program I can run with grbl to operate my router. 

I would try Universal GCode Sender.
https://winder.github.io/ugs_website/guide/platform

Rick23n, 2019-03-23:
Hi  Stuart
I have installed UGS as you suggested and it works fine for me. I have also installed 'FreeCad" for designing but it is going to take me a long time before I will be confident enough to use it. I am still waiting on a response from Banggood in regard to my 16 watt laser. It appears to only have the power of a 2 watt laser. I have tried all configurations with different controllers but all I can engrave is wood and acrylic with power set at 255 and speed around 200 mpm. Not what I expected. 
The first 3 axis Controller that did not work seems that I'm not the only one that has a problem with it. Garthanstey posted the same problem that you commented on 15th Jan. PierricG posted a solution but that did not work for me, the laser still will not fire

StuartB4, 2019-03-23:
Hi Rick. 
Glad to see you are getting on ok with universal gcode sender. 
I did see Garthanstey‘s comments and read about the fix provided by PierricG. Shame it didn’t  work for you. 
Also,  I’ve seen a few people complaining about 15 watt lasers only providing about 8 to 10 watts. Thats really annoying when you don’t get what you are paying for. I hope Banggood sort it out for you. 
You mention Freecad, I have tried it in the past but I’m not much of a designer I’m afftaid. I think I’m a bit short on patience when it comes to learning new things, may be I’m just too old now. I wish you luck with it though.

Rick23n, 2019-03-23:
Thank you Stuart, I just hope that I can achieve my goal with the router or if Banggood come up with a solution for the laser, I would much prefer to do it that way as the laser would be far more capable of finer graphics, I will post as I progress(cant wait too long tho 72 next month!)

StuartB4, 2019-03-23:
>  I will post as I progress(cant wait too long tho 72 next month!)

I know what you mean, every little bit of time counts as we get older, seems to me the older we get the quicker it goes. Good luck with it and happy birthday for next month, just in case we don't talk between now and then.

Rick23n, 2019-03-29:
Hi Stuart
Well, Banggood will not do anything to overcome the problem with my 15 watt laser. After providing videos and  photos proving it only has the power of a 2 watt laser they refuse to accept responsibility and offer nothing in compensation. So buyers beware. I will be looking at other suppliers in the future.

StuartB4, 2019-03-29:
Hi Rick.
Oh dear. Sorry to hear that, I know 15 watt lasers are not cheap. 
I've never bought anything from them but I thought they were a good company. 
Not a lot you can do if they won't accept it back.

arkypita, 2019-04-02:
> The 'X' and 'Y' axis control should be ok but how do I bring the 'Z' axis into the"G" code ? I want to be able to engrave 5mm acrylic to a depth of 1 to 2mm

LaserGRBL is designed for XY engraving. Moving Z axes could be done manually, via custom buttons.
For laser cutting thick materials I will add a "custom gcode between passes" in a future version.

pblacq, 2019-04-11:
Hi Rick,
I use laserGrbl for standard CNC work, and yes you need to compensate for the lack of Z control. I do this with just typing in GCode to get my Z-Axis setup when setting the WCS for the job.  The cool custom macro buttons can also be used, but they will not give you the variability in step size.

closed 2019-11-03 by arkypita