arkypita/LaserGRBL #600: Laser burns in wrong way creating unwanted effect

Elvis999IT, 2019-03-07:
Hi
Since this morning before update to 3.0.9 everithing was perfect. I was at 3.0.4.
Then i begin to have a strange effect, i tried different file and also differen format, different material,
the image it show 4 line diagonal but i have the same effect with orizzontal.
I tried everithin, uninstall th 3.0.9, install again 3.0.4. checked power line, signal, everithing but no result.
[image]

gisep, 2019-03-07:
are you talking about the "low density"?

StuartB4, 2019-03-07:
Whats the actual problem? It’s hard to tell whats wrong because we have not seen how it was before.

Elvis999IT, 2019-03-07:
hi, thanks for the quick answer.
As you can see the top part appear with orizzontal line, one more dark, than one less, again more dark...
It shoul be all the same color.
I also tried with a black square made with Paint, same effect.
It must be just a black block.
Is not the material because before update (or at least is the only difference) i made the same image and was perfect.

Elvis999IT, 2019-03-15:
No one have an idea of what it can be?
i tried different PC, different LaserGRBL version, reflashed arduino...
I Still Have this problem.
Next it will be replace the arduino board...

arkypita, 2019-03-15:
You can exclude LaserGRBL issue. IMHO your is an hardware problem.
Look like "elastic" pull-spring tension, or eccentric problem of your engine/gears that cause laser head accelerating-decelerating during engraving

closed 2019-11-03 by arkypita