arkypita/LaserGRBL #601: Real 15W diode?

gisep, 2019-03-07:
Hi all!
I'm studying my chinese laser driver attached to the "15W" chinese laser diode.
I would understand the effective current it gives to the diode, to understand its real power and try to recycle it to a new laser diode.
I precise that my laser diode is damaged: it will probably affect the results. Please confirm me someone that knows!

I took the following measures on this "15W" (chinese) laser driver. I'm sorry I couldn't measure Amperes at 100%power; that's because my tester has a limit at 200mA!

-I set the laserGRBL power from 0 to 1000: for example, s100 is 10% power.

-With laser diode DETACHED, i measured on the laser "+-" PIN these V:
Power    [%]         V
s1          0,1%     0,7
s10        1   %    0,75
s50        5   %    1
s100     10  %    1,35
s200     20  %    2
s300     30  %    2,65
s1000  100 %    6,6

-With laser diode DETACHED, i measured on the laser "+-" PIN these mA:
Power     [%]        mA
s1          0,1%      27,4
s10        1   %      46,4
s20        2   %      70
s30        3  %     103,2
s50        5  %     162,5

-With laser diode ATTACHED, i measured in series with the laser these mA:
Power     [%]        mA
s1          0,1%        7,5
s10        1   %      17,3
s20        2   %      22,1
s30        3   %      26
s50        5   %      41,2
s100     10  %      81,2
s200     20  %    158,7

I never took electronic lessons, so I don't know what this means.

About V:
I can imagine that 0,7V is something like the minimum; than it increase of about 0,65V each 10%, arriving at 6,5-6,6V at 100%.

About A:
I don't know why the difference between laser attached and detached; maybe the one absorbed is the real "laser power"? (maybe not)

With laser detached, i assume 27,4mA being the minimum; than it increases of about 25-30mA every 1%. Maybe the 100% power will be 2,5-3A?

NOW
6,6V*2,5A=16,5W
6,6V*3  A=19,8W

Why? Is this normal that it overpasses the declared 15W or it is due to "non linear" increasing of A? I could expect it would be 10W or 5W or 3W, but surely not 20W!

About the considerations on measurements with laser ATTACHED:
-0,1% power (s1): mA absorbed = 27,4 - 7,5 = circa 20mA absorbed by the laser; the rest goes in the tester;
-1% power (s10): 46,4 - 17,3 = circa 29mA absorbed
-5% power (s50): 162,5 - 41,2 = circa 121,3mA absorbed

I'm lost!

I think I've reached too many errors in my argument. Can you help me? Thaaaaanks guys :)

gisep, 2019-03-08:
[image]
These grafics could offer a visual support to compare the measures

arkypita, 2019-03-08:
All the measurements you have made have no practical sense.

Current measurements made with tester are not reliable at S100-10% because to obtain power modulation the current is pulsed, while the tester is made for continuous current measurement (or at the limit, alternating current 50hz). The only reliable measurement you can do is when power is 100% (in this case the current is not pulsed, and tester can measure it).

Secondary, measure of current without load take no sense at all, because tester act as shortcut when used to measure current, and I think that the driver is not designed to work on a short-cut. Diode (or at least a low value resistor) is needed to put the driver in condition to provide the designed current.

arkypita, 2019-03-08:
Finally using a damaged diode to measure current on it, is like measuring current flow inside a broken bulb lamp.

gisep, 2019-03-08:
Thank you for your opinion arkypita!

I know there are strong limits in these "no practical sense" measurements, and i thank you for telling me some basics.

About the pulsed power, this surely can't be a precise measure (in the tester the results are moving among near intervals), but at the same time i think it could make a rough idea of quantities: the graphic is almost linear.

About the current with and without load, i'm totally ignorant in this, so now i can imagine (still without well understanding) that those are not relevant for reaching my scope. The same with the broken laser.

At this point my question is: how would it be if it wasn't broken? more current than the "red line" but less than the "blue line"?

Maybe i'm only loosing my time and stealing yours with this experiment with "no practical sense" and i'm sorry to busy this great forum you created, but it was only to better understand something that I normally ignore, maybe uselessly, maybe not.

closed 2019-03-13 by gisep