arkypita/LaserGRBL #602: Laser cooldown / rest

brakthehun, 2019-03-08:
Ok,, so I've dealt with 50+ laser modules by now. I've had a recent string of 1/2 watt UV lasers (7 of them so far) that do great on all these little test runs. I go to do an eight hour engraving and the laser dies about 20 minutes in. Seems like it's the actual diode that dies. 3 different suppliers so far. 

What they say is let the laser sit for five minutes every 30 minutes of running. doesn't account for the 20 minute lifetime of the recent lasers. 

My question is,, can we add an auto pause to the program?? Is there an elegant way to do it??

arkypita, 2019-03-08:
Laser diode has 10-20.000 hrs lifetime if properly handled.

Sorry for your diode, in my opinion you should provide a better heatsink and good thermal contact between ld and heatsink, and-or regulate driving current to a lower value.

It is very common that china seller, sells ld telling a ovverrated value of power (ie 2,5W for a 2W). Always check datasheet of diode

arkypita, 2019-03-08:
About autopause i think that the most elegant way is to put it outside of gcode, i mean in lasergrbl itself, and send pause/resume.

I think is already in dev roadmap

m3vuv, 2019-05-16:
i had the same issues in m3 mode,after doing a 20min run with a lot of on off cycles my diodes would die,i was told my driver needed a soft start and stop for the diode,my driver had soft start but not stop,i put a 0.22ufd cap in parelell with the diode to turn it off slower,im led to belive the thermal shock of turning on and off without a soft on and off kills them,after my mod it seems fine,im using a nichia m-140 runing at 1100ma.

closed 2019-11-03 by arkypita