arkypita/LaserGRBL #603: working table starts to move in y immediately on start up.

tjensen03, 2019-03-12:
Hello,
I'm using the SainSmart 5.5W laser module (for the first time yesterday). When I start LaserGRBL, the table automatically starts to move in -y direction. The only way to stop it is to kill the power and unplug the USB cable.  I did manage to get it to connect and run ok after a few tries. But tonight, no luck... On connection, it starts to move .
If I use Candle, (the CNC software that came with the machine) as a test of the system, there is no issue. Everything seems fine.
Any idea what's causing this?
My other issue is with the PWM mode... it isn't strong enough to burn. I have to use the "power" switch on the PCB to get any burn at all. From what I read, this might be the S32=1 (from 0) issue? 

Regards 
To,

tjensen03, 2019-03-12:
Hi. Thanks for the quick response.
The first thing I tried was to uncheck the resets. I am currently using an old laptop running XP. I will load the software onto a Win 7 laptop and try that today.

tjensen03, 2019-03-12:
SOOOOO close!
OK, I have the laser pulsing and burning now. Things move in the correct directions. Still working on bland and white images, haven't moved on to grey scale yet. Last remaining question... how do I set the home position? 
Thanks!!!

arkypita, 2019-03-13:
The button with a globe and pin set zero position, as its tooltip say.

tjensen03, 2019-03-13:
Great. Thank you for your help an patience!

StuartB4, 2019-03-13:
> but two options that might be useful... a low power "laser on" setting to allow focus, but not strong enough to engrave.

Laser on to focus.

[image]

>  And a preview area command that would move the laser around the work area at low power so that the user can ensure the work piece is properly positioned.

Make another Custom button this time just leave the Type setting to read Button.

Add this code to the GCode box.
M4 
S2
F1700
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom] 
G0 X0 Y0 F4000
M5

Then add a Tooltip again.
Select Connected again and click save.

StuartB4, 2019-03-13:
Another button you might find useful is to find the center of the image you have prepared.

[image]

tjensen03, 2019-03-15:
Thank you Stuart!!! That is exactly what I needed!!!

StuartB4, 2019-03-15:
Glad to help. :)

tjensen03, 2019-03-16:
Hey Stuart, I tried running the focus command and it doesn't seem to work. Laser doesn't come on even with $32=100 and it throws an error on second click "Expected value is..."  The center and work area commands are fine.
Any thoughts?

StuartB4, 2019-03-16:
What version of GRBL are you using? It should tell you here:

[image]

Make sure you typed $32=0 and $32=1 not S32=0 it should be the dollar sign not S.

[image]

tjensen03, 2019-03-16:
Yep... figured it out... S instead of $. Old glasses and older eyes don't mix well LOL!

Again, thanks for your help. This is exactly the setup I wanted!

StuartB4, 2019-03-16:
Haha, same here, old eyes, easy mistake to make, glad you sorted it.

closed 2019-04-02 by arkypita