arkypita/LaserGRBL #604: Line spacing and off measurements.

Mk5mod0, 2019-03-16:
Hello

I just got a laser burner and am using LaserGRBL with it and love it.
But after my first controller burned up (The controller that came with the machine ) I replaced it with a new Arduino Nano. After flashing the Arduino with Arduino IDE I thought I was good to go. But when I go to burn something there are large spaces between the lines, the program messes up the picture completely, or the measurements are off. Please help.

[image]
[image]
[image]

arkypita, 2019-04-02:
https://github.com/arkypita/LaserGRBL/issues/607

Mk5mod0, 2019-04-07:
Thank you for all of your help everything is up and running like new.

closed 2019-04-07 by Mk5mod0