arkypita/LaserGRBL #606: My Custom buttons Zup and Zdown don´t work in new version

David7820, 2019-03-20:
Hi, my customs buttons Z up and  Z down don't work after loading the new version.
step up: $J= G91 Z[jogstep] F[jogspeed]
step down: $J= G91 Z[0-jogstep] F[jogspeed]
However, the new jog "home" icon turn down the z axis when make homing.
I do not know if it's just my case or it's happened to more people.
Thanks.

arkypita, 2019-03-20:
Yes, it's the same cause of #605
The new jog control loose the slider for jog step and broke some features.

I suggest to put fixed value instead of jogstep and speed in your variables until i have found a good way to put both features together

David7820, 2019-03-20:
Ok, Thanks.
Você é grande.

arkypita, 2019-05-07:
Whould you like to try? https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.11

David7820, 2019-05-07:
Thanks,  I'll try tomorow.

arkypita, 2019-05-08:
fixed

David7820, 2019-05-14:
Ok, works fine, like before.Thanks.

closed 2019-05-15 by arkypita