arkypita/LaserGRBL #607: Size change after update

chriswalker247, 2019-03-22:
Hi hope you can halp! :)

using eleksmaker a3 pro. Used it under the 0.9 grbl fine and a prev version of lasergrbl. Creating 100mm x 100mm images on 3mm ply with no bother in lasergrbl.

Updated grbl to 1.1g from 0.9 and updated to latest lasergrbl software (3.0.10). Now it's like it's translating 1mm to 1cm or something similar. When I trace the image area it goes to the edge of the machine causing it to bump the edges. 

not sure which configs have changed for it to cause that. thoughts?

StuartB4, 2019-03-22:
Sounds like it might be a microstepping issue.
Have a look here for how to set them correctly
https://www.youtube.com/watch?v=z3bUg9InVeA

chriswalker247, 2019-03-22:
Watched the video and used your spreadsheet. Nailed it. So thank you! :) just need to speed up the engraving speed now and sorted. thanks again.

StuartB4, 2019-03-22:
Thats good to know, happy to help.
It's not my spreadsheet, it was created by Charley Thomas, who was kind enough to make it available to anyone that wanted it, but it is handy to have.

[image]

closed 2019-04-02 by arkypita