arkypita/LaserGRBL #608: Right slider by jogging control missing

ShKh001, 2019-03-22:
Hello, with the new update the right slider by jogging control is missing.
Thanks for your help.
[image]

Stankovo, 2019-03-31:
I have the same problem! Help!Still, low speed at burning out except vector. How to fix?

Stankovo, 2019-03-31:
Went back to 3.09 appeared as before right regulator.

StuartB4, 2019-04-01:
Read Arkypitas reply comment here.
https://github.com/arkypita/LaserGRBL/issues/606

Stankovo, 2019-04-01:
I suggest to add a button that you hold down the mouse, you can move in any direction like a joystick. Pulled to the right a little - moving slowly to the right. I pulled harder, moving harder. Pulled right up - will move in this direction. Joystick button. He pointed the mouse, pressed the left button and holding adjust the direction and speed of the laser head.

arkypita, 2019-04-02:
> I suggest to add a button that you hold down the mouse, you can move in any direction like a joystick. Pulled to the right a little - moving slowly to the right. I pulled harder, moving harder. [...]

This is not possible with the actual jogging protocol supplied by grbl/gcode.
You cannot send "runtime" jogging but only "go there with this speed" commands

arkypita, 2019-05-08:
fixed

closed 2019-05-08 by arkypita