arkypita/LaserGRBL #610: In v3.0.4, the X axis is moved uniformly in the Laser Mode, in version 3.0.10 X-axis moves as if not seeing that the mills in Laser Mode

sbrv, 2019-04-01:
In v3.0.4, the X axis is moved uniformly in the Laser Mode, in version 3.0.10 X-axis moves as if not seeing that the mills in Laser Mode

arkypita, 2019-04-02:
the request is not understandable

closed 2019-04-02 by arkypita

StuartB4, 2019-04-02:
Do you mean it's not moving in millimeters ?
If so then it may be the steps per/mm setting in GRBL settings has changed in the 3.0.10.
See my first comment here: https://github.com/arkypita/LaserGRBL/issues/607

sbrv, 2019-04-02:
I mean, in version 3.0.4, the X-axis with the laser moves all the time at a constant speed, and after upgrading to 3.0.10, the X-axis starts to move unevenly: accelerates and slows down, as if the "laser mode" parameter is 0. As a result, the picture quality has deteriorated dramatically.