arkypita/LaserGRBL #611: No movement/ power.

Mooftah, 2019-04-02:
assembled Tsemy Direct DIY 3040, Led light on, and connectivity to GRBL 3.0.1. I cannot move the engraver manually, there is no laser on, no laser fans, and when i tell the program to run the project...it completes with zero movement from the machine.

arkypita, 2019-04-02:
Did you connect DC 12V power supply?

Mooftah, 2019-04-02:
Yes.. I even tried a different one. Still nothing.

arkypita, 2019-04-02:
The problem is only with LaserGRBL or also with others gcode sender like UGS etc?

Mooftah, 2019-04-02:
First off. thank  you so much for being so responsive. you should know im very new at this GRBL game so my trouble shooting skills are limited. i agree think its a power issue not a software issue. Not sure what UGS is?

arkypita, 2019-04-02:
Isba software called "universal gcode sender"

closed 2019-11-03 by arkypita