arkypita/LaserGRBL #612: Help with selecting a new controller board

BruceyZG, 2019-04-03:
A year or two ago I bought a noname laser engraver kit from China. The hardware was excellent, but the controller board was a cheap EleksMaker clone. It kind of worked, but had problems the whole time. Now it's close to failing completely; I can't even flash new firmware onto the Arduino without pulling it out of the rest, so I decided to get a completely new board. But instead of just blindly buying the first one that seems good enough, I wanted to ask for your opinion.

This is the old setup:
[image]

Notice how the green board (I don't know what to call it) is connected to the controller by only 2 wires. All the newer, similar boards (EleksMaker Mana SE and so on) have the 3 pin layout (12V/GND/PWM), whereas mine only has +/-
The 2 pairs of wires on the right of the board go to the laser and fan. Also notice the 3rd connector below the laser and fan wires which says TTL.

To keep it as short as possible, my goals are:
- Getting the laser engraver to work again
- Being able to switch between full power and weak mode (didn't work with the old setup), possibly by hardware switch/button
- Engrave grayscale

I hope you guys can give me some good suggestions! If I need to buy more than one circuit board and throw the old ones away, that's also ok. If you need more data/photos, just let me know.

Cheers, Brucey

dungjk, 2019-05-09:
My opinion
1 x Arduino UNO + 1 x CNC Shield + 2 x DRV8825
It's cheap yet stable, easy to find, easy to assembly and ton of document online & grbl wiki.

StuartB4, 2019-05-09:
It looks like the setup you have is not capable of doing greyscale because the TTL on the laser driver board (the green one) is not connected to the Arduino (the blue board) through the EleksMaker Mana SE controller board. 
You would need a new controller board instead of the EleksMaker Mana SE that has a TTL/PWM connector.
Then you would connect the TTL/PWM from the laser driver board (green one) to the TTL/PWM on the new controller board. That would then connect to the Arduino's PWM output pin and allow laser modulation/power setting/ and greyscale burning.

As BruceyZG said, an Arduino UNO R3 (I would go for a genuine one, no usb driver problems), also get the plug in chip version, not the surface mount one, if you do damage the Arduino you only have to buy a new chip and not a new Arduino.

A Protoneer CNC Shield (Latest version 3.51) and 2 DRV8825's is a good choice, although you could still use the 2 little red motor driver boards from your Eleksmana board instead of the buying new DRV's.

If you buy a Protoneer shield from Ebay you have to assemble it yourself, solder in loads of little pins, or you can get one fully assembled from their other supplier https://www.elecrow.com/arduino-cnc-shield-v3-51-grbl-v0-9-compatible-uses-pololu-drivers.html

It says it's GRBL 0.9 compatible but it is also compatible with GRBL 1.1

Have a look on their website forum for more info and help.
https://forum.protoneer.co.nz/index.php

StuartB4, 2019-05-09:
[image]

closed 2019-11-03 by arkypita