arkypita/LaserGRBL #619: Z-Axis Control

pblacq, 2019-04-11:
First off, awesome application.  Nice clean UI and most importantly for me, good webSocket support.

I apologies about repeating this request/question...I am just hoping I can add a little more justification for Z-Control buttons.

I have a dual purpose machine, supports both router and laser.  It also has a large cutting bed for the router and as such needs webSockets, or very long cables (EMI, SNR, etc problems)...

I love this apps accuracy and reliability over WiFi, and have not found an alternative.  But the lack of Z-Axis control (yes I do use custom buttons, well will do again when fixed), would be really appreciated.  

The new position step-size aware jog buttons are nice, however could you consider adding a simple step size up/down control 0.1->1->10 and back again so really fine tuning of positions can be achieved

Thanks again for your app.

arkypita, 2019-04-27:
This is my number one priority together with custom gcode header and footer and between passes (where you can move Z between each pass) but it is a busy period for me.

pblacq, 2019-04-28:
Thanks for the update.  Look forward to it, and appreciate that this is hard to fit in around making a living.

arkypita, 2019-05-07:
Whould you like to try? https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.11

pblacq, 2019-05-10:
Thanks, this enhancement works very nicely, much appreciated.

pblacq, 2019-05-13:
Hi, Sorry for reopening the issue.

I initially tested this with my Z-Axis lying on the workbench, and did not notice that the movement was inverted.  Now assembled I notice that clicking up, results in the Z moving down,  Closer inspection of the Gcode sent confirms that the new Z jog controls are inverted.

arkypita, 2019-05-13:
What document determines which is the correct orientation of an axis?

arkypita, 2019-05-13:
I mean, what can confirm that is not your axis following a wrong convention, instead of LaserGRBL?

arkypita, 2019-05-13:
Look that CNC convention (that is a convention, not a rule) use Z up positive, and Z down negative.

You are right. In my LaserGRBL i choose to send positive increment to go down, because seems to make more sense to me, but apparently the "standard" is the opposite.

I think I will invert it ASAP.

pblacq, 2019-05-13:
Correct it is a convention that up is away from work-surface...not in a documented standard to my knowledge. For example: home is in the plus x,y,z direction (which was the wrong way around in my head to start with)..

closed 2019-05-15 by arkypita

hariantody, 2019-05-16:
dear master Arkypita
how to change output d11 (laser on / off) can run micro servo (down / up) to write with pen ........?

pblacq, 2019-05-16:
> dear master Arkypita
> how to change output d11 (laser on / off) can run micro servo (down / up) to write with pen ........?

Hi Hariantody, I suggest you repost this question as it is currently in a closed enhancement thread.  My thoughts to your question are that you will need some electronics and to modify Grbl to make D11 output a suitable PWM signal to control a servo motor, rather than this being something laserGrbl can do.

hariantody, 2019-05-16:
dear master pblacq
thank you for the suggestion ....... good luck always

abatinxs, 2019-05-23:
hariantody You must put to atmega firmware GRBL with PWM signal for servo (1,5ms neutral), and use LaserGRBL like me :) I showed it in my YT https://www.youtube.com/watch?v=7ZpJGVCwD4o
Modified GRBL You find in github. It`s all  
Best regards from Poland.

arkypita, 2019-05-23:
That's awesome! I want one :-)

abatinxs, 2019-05-24:
Arkypita try this - https://github.com/robottini/grbl-servo

hariantody, 2019-06-01:
Dear Abatinx
first I tried uploading grbl 1.1 servo hex to no avail. both I uploaded grbl 9.0i servo hex and unchecked pwm support and succeeded ... the servo moves like the laser command.thank you for the instructions
greetings from Indonesia