arkypita/LaserGRBL #620: CNC 3018 2 IN 1 + LASER 10000mW

EBEALIEN, 2019-04-11:
Hi everyone,
I've already written in the other topics, and being closed I didn't want to disturb.
A month ago I purchased the 3018 2 in 1 cnc with 5500 mW laser, the problems were born,
until I returned the equipment and they repaid me (amazon).
Yesterday another identical one arrived from China, but with 10000mW laser, same exact story even with different card, the same problems.
At power on the laser continues to burn even if in pwm, and without having loaded anything, and does not detach when it has to detach. $ 32 is = 1
Help me, I already spent 700 euros to see a laser simply work. Please a good code and try doing a video. Otherwise I also return this.
Thanks friends, Thanks. Antonio Italy  :-(

EBEALIEN, 2019-04-16:
**Update**, 
if I try to use lasergrbl after using something else, my equipment is probably confusing.
I've tried a few solutions, but I'm convinced that **laser grbl** is the best, and I want to use this.
If I turn on the PC and my equipment, the laser in PWM works, I have to figure out how to set it to cut, and how to set it to take a picture.
I'm not an expert, but I would like to understand how to dose the power to cut a piece of cardboard and how to cut 3 mm plywood.
The laser as mentioned is 10000mw and I don't understand how to dose it.
Step by step we find the solution.
thanks to who helps me.
Antonio - Italy

arkypita, 2019-04-16:
Ciao Antonio. Ti aiuterei volentieri ma in questi giorni sono troppo occupato.
Scrivimi a info@bergamo3.it e descrivimi il problema, magari ci sentiamo x tel dopo la Pasqua

StuartB4, 2019-04-16:
When you have prepared your image for burning you will get to the Target Image settings.
To cut out it is better if you change the M4 to M3. 
Also you can change the Engraving speed in the Speed setting above it.
[image]

For cutting thicker materials you will probably have to do more than 1 pass of the laser.
If you have a Z Axis laser that you can move up and down, you can also move it down a very small amount after each pass.
[image]

EBEALIEN, 2019-04-16:
Thank you guys,
whether or not the laser goes, in this site there is professionalism and seriousness.
Answer without assuming that someone has to understand everything, and this, is a merit, for men, and is more important than anything !. 
Now, I read the whole site, and searched the topics, well tonight thanks to **StuartB4, arkypita,** I try the cutting pass.
Unfortunately I can't find a single file to try already good, and I, among a thousand "youtuber Italians" (which are fake)
I don't understand anything, only by foreigners can I understand something. I will keep you updated and meanwhile support the project because it deserves it!
Antonio

EBEALIEN, 2019-04-16:
updating:
I tried the solutions of the video with "tree" different from 80 x 80 mm,
and starting from the original settings, I tried to raise and lower, but, I get the same results, only the speed changes and I know that at most it reaches 1000.
Now I think the ideal (I haven't found the means to compare),  it would have a test file and another test file, to verify the differences in power, and all tested and compared by those who post it.
I mean, if they have differences I must have me too, if I did not I think of a problem. I have a few days to give OUT the successful or unsuccessful closing of the order.
[image]
in practice I can't create a simple one, to see the various powers,
I saw a 4 mm plywood cut, with a 2500 mw and so if mine has 10000 mw ............................. .

EBEALIEN, 2019-04-17:
I spent a star on wood, with 3 passes at 600 mm / min ---- then I put in place of 0 - 255 I tried to put 400 - 600, and on the main page I decreased the 2 speeds to 50%, and turned up the power to maximum.
_this is an example_
[image]
[image]
But how do I understand the minimum power and up to the maximum, and then I have to keep 0 - 255, or can I insert the higher plane values?
_Grazie StuartB4 and arkypita_

arkypita, 2019-04-27:
In grbl configuration you can set.

Settings and sample values | Description
-- | --
$30=1000. | Max spindle speed, RPM
-- | --
$31=0. | Min spindle speed, RPM

Both values are totally arbitrary.

When you send S commands they will be proportionally scaled to PWM power (that has 8bit physical range of 0-255) using $30 and $31 as values for scaling S command values.

For example:

With $30=1000, $31=0 an S500 will correspond of a PWM value of 128 that is 50% of power.
With $30=10000, $31=0 an S500 will correspond of a PWM value of 13 that is 5% of power.
With $30=255, $31=0 an S128 will correspond of a PWM value of 128 that is 50% of power.

Nothing prevents you to put a totally arbitrary value but I suggest only two different kind of setup:

- 1000/0 if you want to express S power in per thousandth values where S500=50.0% and S320=32.0% etc
- 255/0 if you want natural conversion between 0-255 bitmap scale of gray and 0-255 natural 8bit PWM values.

arkypita, 2019-04-27:
Later in LaserGRBL in the raster import procedure you can put two others different S-MIN and S-MAX that has a totally different functions. They control how grayscale values are converted to S values: white became S-MIN, black became S-MAX

Summarizing:
$30 and $31 control conversion between S values and PWM power.
S-Min and S-Max control conversion between grayscale image and S values.

Usually they are bot settled to the same values (ie 0-255) so a white pixel will be 0% of power, a 50% gray became 50% of power and so on to black pixel 100% of power but there are some situation where you could be advantaged using different values.

For example if you have an high power laser and you want to use only a portion of the power you can set lower S-Max in raster import to produce lower S Values that will be engraved using not all the power.

Another situation is if your laser power is too weak to engrave light tones, or does not fire at all at lower values, so you can "pump-up" the power settings S-Min to a higher value so the "white pixel" are converted for example to S20 instead of S0 and so on.

EBEALIEN, 2019-04-29:
Well, I understood well!
The value scale, I can adapt it to my needs, but in the end it will return 0-255, which is set in thousands, or in percentages, and I understood them.
I will do many tests to understand even more, and I will report them.
What I don't understand (having no reference points), is, to estimate the power of my 10 w laser.
I see 2500 mw lasers identical to mine, make 3 pass cuts on 4 mm, and I with my 10 mw can do the same, but with difficulty.
But in the end, the maximum lift beyond which it does not go,   how can i test it physically ??
_@arkypita Whenever possible if possible we can hear each other! I have sent mail with video, but it returns an error.
@arkypita Quando vuoi se possibile, possiamo sentirci!_
In essence, understanding the maximum power would be ideal, and then work with percentages.
For now we say that the laser works well with the software, 
and thanks to you work well, I made a small photo, and some cuts. When I turn on the equipment it scans 3 times quickly and then waits for work.
[image]
before I passed a png vectorized 600 speed 0-255. then I cut the svg 300 speed and put 700 - 701. sigh

arkypita, 2019-04-29:
> What I don't understand (having no reference points), is, to estimate the power of my 10 w laser.
> I see 2500 mw lasers identical to mine, make 3 pass cuts on 4 mm, and I with my 10 mw can do the same, but with difficulty.

Without a dedicated instrument, or without having a laser diode to compare it is difficult to understand the real power of a laser. You only can do some empiric consideration or evaluate voltage/current consumption.

Keep in mind that focusing is very important when cutting/engraving: a diode of 2W focused on a squared dot of 0.1x0.1mm (0.01mm^2) means a power per surface unit of 200W/mm^2, while a 10W diode focused on a dot of 1x1mm (1mm^2) means a power of 10W/mm^2.

Also please note that focusing depends from lens quality and in general is more difficult to obtain small focus dot with higher laser power.

Focusing is done by increasing/decreasing distance from laser to the surface using Z axis control or by 
rotating focus lens gear. Be care to your eye when evaluating focus. Evaluate focus using the lower power available to your laser to fire-up (i.e laser mode off, S30 of 255 power, M3 to turn on the laser) and please wear a good pair of safety glasses.

Some hint here: https://jtechphotonics.com/?p=2602

peagerfw, 2019-04-30:
If i have 15w laser how to set maximum?
sorry i am idiot with this setting :)

closed 2019-11-03 by arkypita