arkypita/LaserGRBL #622: Can't Connect Anymore

poppen03, 2019-04-20:
Newly arrived engraver. LaserGRBL connected and worked fine. Now it can't connect. It says "connecting" then "disconnected" 10 seconds later and a soft click can be heard. Screenshot attached.  I can't figure out what has changed. The laser itself seems to work though there is no control over intensity, just an on/off switch on the board. Any ideas?
[image]

StuartB4, 2019-04-20:
Are you sure the com port you are trying to connect to is the correct one.
Try changing it for a different one with the drop down box underneath the File menu.
Or check in Device Manager in Windows to see which one your laser machine/Arduino should be on.
Have you tried a different usb port or a different usb lead.
Did you have to install a driver when you first set up the machine, is so try re-installing it.

poppen03, 2019-04-21:
Thanks for the suggestions. I've tried all of them and tried two different computers. Still no success.

arkypita, 2019-04-21:
As can be seen in picture you post, your hardware use benbox firmware.

As clearly written in http://lasergrbl.com/faq LaserGRBL is only compatible with GRBL firmware. Please read

StuartB4, 2019-04-21:
> Newly arrived engraver. LaserGRBL connected and worked fine. 

Thats strange, you say it worked and now it does not. But it should not work, are you sure it was LaserGRBL that worked.

poppen03, 2019-04-21:
It connects to Benbox but still doesn't work right. I'm thinking that maybe I altered the firmware when playing around with Benbox. I'm going to try and re-flash the board with GRBL because I prefer to use LaserGRBL. I'll let you know how it goes.

StuartB4, 2019-04-21:
Ok poppen03. Good luck with it.

poppen03, 2019-04-21:
OK, that worked. Only problem now is I'm not getting burning power (only 4.5v) to the laser.

poppen03, 2019-04-22:
Thanks for the suggestions.  I've tried all of them except replacing the USB lead (don't have one yet).  I have also tried running it on 2 different computers without success.

closed 2019-11-03 by arkypita