arkypita/LaserGRBL #624: Engraving on Curved Surface

rynaldi95, 2019-04-22:
Hi,
I am Rynaldi. Currently, I am working on a laser engraver machine that will engrave some pictures on a longboard. However, the surface of the longboard is not planar. There are some curved areas on it. Do you have any suggestion on how to do it? Because I think with different surface height, I have either change the focus of the laser or adjust the height of the laser module by controlling a Z-axis. Thank you in advanced :)

arkypita, 2019-04-27:
If the curves are not so tall i think no problem but if there are more then 5-6mm the out of focus could be a problem.

I think you need a 3 axis head and a more advanced software that can do a cycle of probe/map of the longboard surface and correct/adjust the Z during engraving to follow your surface.

arkypita, 2019-04-27:
Something like this https://github.com/martin2250/OpenCNCPilot

rynaldi95, 2019-04-30:
I'm sorry if I'm asking to much. But is it possible actually to combine your software and the OpenCNCPilot? I mean to combine the surface scanning functionality from OpenCNCPilot with LaserGRBL. If yes, could you let me know how should I start or which part of your code that I should edit? Thank you.

arkypita, 2019-04-30:
I think merge both software is not possible because they are very different.
What you can do is to generate GCode using LaserGRBL and export saving to file, then load in OpenCNCPilot and use this one to stream it using its autolevelling feature.

e6910, 2019-05-04:
I theory at least, you can do it with a rotary attachment connected to the Y axis. If the stepper for the attchment is the same as the y stepper no further modificatios are (supposedly) needed. 

I DON'T HAVE ONE, but I'm planning to make/purchase one

closed 2019-11-03 by arkypita