arkypita/LaserGRBL #625: No Laser Control

poppen03, 2019-04-24:
I just got a Chinese 3w laser engraver kit.  I flashed the board with GRBL 1.1 and have control over the motors. The laser turns on at a low power and stays on. I have tried checking and unchecking the PWM box, changed the laser enable  mode in the configuration from 0 to 1, the spindle max to 255 and spindle min to 0 and tried sending individual M3, M4 and M5 commands. Nothing has any effect.

What else can I do to get the laser to respond to commands?

StuartB4, 2019-04-24:
Is it a new machine or a used one.
Does the board look like this one with the Blue Arduino Nano plugged in.
[image]

Some older boards have a jumper underneath the Arduino Nano board that needs moving across to make it compatible with GRBL 1.1

Switch the machine off before removing the Nano to find out.
The newer versions no longer have the jumper.

poppen03, 2019-04-24:
Yes, it looks exactly like your photo. I bought it new. What do you mean by "needs moving across"?

StuartB4, 2019-04-24:
I'm just going by what I found on the net, which was this.

> 1. unplug your machine.
2. Remove the nano (piece you plug the usb into) from the motor board
3. Underneath, there should be a jumper and 3 pins. Move the jumper to short out the center and other pin (that was not under the jumper already)
4. confirm your firmware (0.9 or 1.1)

It just pulls off and you move it to the middle pin and the other unused pin.
But looking at the image of the board I posted it looks like that jumper is a voltage
selector, so I don't know how changing the jumper would help.

poppen03, 2019-04-25:
Thanks, but there is no jumper at all. Are these (see photo) the contacts you are talking about? If so, which two need to be connected?

[image]

poppen03, 2019-04-25:
[image]
[image]

StuartB4, 2019-04-25:
Does yours have a version number printed on it, it actually looks older than the one in the image I uploaded, it has through hole resistors and a stardard led compared to the surface mount resistors and led.
Without knowing what the holes are for I would'nt connect anything there.

poppen03, 2019-04-25:
Agreed! This photo shows the only thing that might identify the board. I'll see if I can get it translated. I'm starting to think I should abandon this board and use an Uno with a CNC sheild.
[image]

StuartB4, 2019-04-25:
You could have a look here first:
http://benboxlaser.us/index.php
There are people who swapped from the Benbox firmware system to GRBL on their machine.

poppen03, 2019-04-25:
Thanks, I really appreciate your prompt replies and help. Unfortunately i didn't find any help. Might switching to Grbl 0.9 help?

lloyd1941, 2019-04-28:
Gyday Poppy
that board is out of date. send for a newie?

closed 2019-11-03 by arkypita