arkypita/LaserGRBL #626: Minimum print size

lodale, 2019-04-25:
Hi
I have tried to vectorize a raster file (png) to a printable vectorized file. In the target image

lodale, 2019-04-25:
Hi again ( I accidently closed the comment). I am running Lasergrbl ver. 3.0.10 (just updated). I try to vectorize a raster fil, in the target image I could not put in the right L and W. The values I want are L= 5.2 mm and W=1.3 mm. When i put these values in the target image window they change to red. and do not show up in the "print window".  How could I adjust LaserGRBL to be able to print images at small sizes, the bigger ones work
[image]

lodale, 2019-04-25:
Oh very embarresd i put in values like cm and not mm,

closed 2019-04-25 by lodale

StuartB4, 2019-04-25:
Easy mistake to make. At least you realised.