arkypita/LaserGRBL #627: Please help connect 15w laser

peagerfw, 2019-04-26:
Hi all,please help me connect my 15w laser to CNC 3018
Sorry for my English :)
My board is woodpecker 3.2
updated to 1.1f
Laser is 15w from acan
[image]

first i must changed +12v and PWM but on laser is different position.

[image]

My problem is:
after connect to board Laser burn if its LASER GRBL program open or close
all the time is on connector +12V,PWM 7.82V
[image]

My setting is now
[image]

I dont know how to set. its possible ?
Please help :(

peagerfw, 2019-04-28:
I helped myself :/
PWM from orig. laser connection was not good for my laser and i used PWM from board
[image]

My question is
1) how change frequency for my laser (i need 9hkz)
2) after change frequency is this 9hkz on (+12v,GND) laser connector too but only spindle motor?

[image]

My laser
Output wavelength: 445nm~450nm
Output Power: 15W ,average 8W
TTL frequency:0V(OFF) 5V(ON),PWM frequency:<9Khz
Laser light:Blue Violet Light
Working voltage: DC=12V
Working current: I<3A

arkypita, 2019-04-28:
> My question is
> 
> 1. how change frequency for my laser (i need 9hkz)
> 2. after change frequency is this 9hkz on (+12v,GND) laser connector too but only spindle motor?
> 

Both question are about grbl.
Write on grbl github page instead of here.

https://github.com/gnea/grbl

StuartB4, 2019-04-28:
I think you can only get to 7.9kHz on grbl. You have to edit the cpu_map. file in the grbl libraries then compile and reupload it to the Arduino. Before you do though make a copy of cpu_map.h and put it somewhere else,  just in case it goes wrong, then you can just replace it, compile and re-upload to the Arduino.

Change PWM Frequency.
comment out line 143 with // and remove the //'s from line 141

138 // Prescaled, 8-bit Fast PWM mode.
139 #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM20) | (1<<WGM21)) // Configures fast PWM mode.
140 // #define SPINDLE_TCCRB_INIT_MASK (1<<CS20) // Disable prescaler -> 62.5kHz
**141 // #define SPINDLE_TCCRB_INIT_MASK (1<<CS21) // 1/8 prescaler -> 7.8kHz (Used in v0.9)**
142 // #define SPINDLE_TCCRB_INIT_MASK ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
**143 #define SPINDLE_TCCRB_INIT_MASK (1<<CS22) // 1/64 prescaler -> 0.98kHz (J-tech laser)**

Compile and re-upload GRBL to 

> Arduino.
[image]

peagerfw, 2019-04-28:
> I think you can only get to 7.8kHz on grbl. You have to edit the cpu_map.h file in the grbl libraries then compile and reupload it to the Arduino. Before you do though make a copy of cpu_map.h and put it somewhere else, just in case it goes wrong, then you can just replace it, compile and re-upload to the Arduino.
> 
> Change PWM Frequency.
> comment out line 143 with // and remove the //'s from line 141
> 
> 138 // Prescaled, 8-bit Fast PWM mode.
> 139 #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM20) | (1<<WGM21)) // Configures fast PWM mode.
> 140 // #define SPINDLE_TCCRB_INIT_MASK (1<<CS20) // Disable prescaler -> 62.5kHz
> **141 // #define SPINDLE_TCCRB_INIT_MASK (1<<CS21) // 1/8 prescaler -> 7.8kHz (Used in v0.9)**
> 142 // #define SPINDLE_TCCRB_INIT_MASK ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
> **143 #define SPINDLE_TCCRB_INIT_MASK (1<<CS22) // 1/64 prescaler -> 0.98kHz (J-tech laser)**
> 
> Compile and re-upload GRBL to
> 
> > Arduino.
> > [image]

Pls how Compile and re-upload GRBL?
which program i must use.Can you write me steps?
Really thank you for your help.

StuartB4, 2019-04-29:
Do you have the Arduino IDE program on your computer.
Did you install GRBL 1.1f with it yourself.

peagerfw, 2019-04-29:
I dont know if understand but iI used Xloader for updated my board

StuartB4, 2019-04-29:
In that case you wont be able to edit the cpu_map.h file.
It is stored in the grbl folder on your hard drive if you updated grbl with the Arduino IDE program.
But you used the .hex file to upload version 1.1f  and you cant edit that the same way.

peagerfw, 2019-04-29:
yes i only upload hex file 1.1f via Xloader to my woodpecker board. Can you write me please how to make,what i need? Sorry i am total noob with this.

arkypita, 2019-04-29:
This is the best guide: https://www.youtube.com/watch?v=m998bYioHqs

arkypita, 2019-04-29:
Video show how to compile and upload directly from grbl source (not via precompiled .hex) using Arduino IDE, so its allow you to edit cpu_map.h and upload your own customized grbl version for specific needs.

StuartB4, 2019-04-29:
@arkypita .
I'm not sure you can do it that way on the Woodpecker board because the Arduino is built on to the controller board. Looks like a bit of a nightmare.
https://github.com/gnea/grbl/issues/336

arkypita, 2019-04-29:
I think that if @peagerfw can update via xloader he can also upload via Arduino IDE

arkypita, 2019-04-29:
Or at least open Arduino output directory and find the .hex produced by compiler, and use xloader to burn it.

peagerfw, 2019-04-29:
good idea... yes i used for flashing to 1.1f Xloader and work it... i check this video and go make it...but i dont understand ... now i must create hex file via Adruino where its posible change Khz for my laser and next step upload hex file to board?

peagerfw, 2019-04-29:
I have downloaded and instaled adruino 1.8.9 software
Next i open adruino,next i include library ,next click add zip library.
Next add grbl master or which grbl libraries i must use for woodpecker?

peagerfw, 2019-04-29:
i have problem with compiler for board arduino nano
[image]

peagerfw, 2019-04-29:
I used only USB and in Arduino i set NOT arduino nano (not write to board with this) but "Arduino Duemilanove or Diecimila" and work it!
I used new GRBL MASTER ist new 1.1g
Where i can check if is my setting 7.8Khz?

StuartB4, 2019-04-29:
To check the 7.8 kHz have a look here:
C:\Users\ **your user name** \Documents\Arduino\libraries\grbl
You should see cpu_map.h file.
Open it with a text editor and check this part.

[image]

peagerfw, 2019-04-30:
Yes i make it,but how to verify if is real changed to 7.8Khz??
Here is my changed cpu.map.h
[image]

peagerfw, 2019-04-30:
OK i compile and sent again and work it :)
Thank you so much for help and your time.

peagerfw, 2019-04-30:
After changed frequency from 1Khz to 8Khz laser burn very diferent.
For me is with 8khz more nice (focus)sharpen but burning is not finish and with 1Khz is all fat 
what is bad? 
[image]

arkypita, 2019-04-30:
Different laser driver works better with 8KHz others with 1KHz.
Mine is good with the default of 8KHz. Never try with 1KHz

There are some thread about PWM freq on gnea/grbl page:
https://github.com/gnea/grbl/issues/55
https://github.com/gnea/grbl/issues/34
https://github.com/gnea/grbl/issues/227

peagerfw, 2019-04-30:
Never try with 1KHz...why???

peagerfw, 2019-05-02:
> Different laser driver works better with 8KHz others with 1KHz.
> Mine is good with the default of 8KHz. Never try with 1KHz
> 
> There are some thread about PWM freq on gnea/grbl page:
> [gnea/grbl#55](https://github.com/gnea/grbl/issues/55)
> [gnea/grbl#34](https://github.com/gnea/grbl/issues/34)
> [gnea/grbl#227](https://github.com/gnea/grbl/issues/227)

How much frequency is possible setting... only 1Khz and 8Khz?

StuartB4, 2019-05-02:
> Never try with 1KHz...why???

I don't think Arkypita meant that you should'nt try it.
I think meant that he's never tried with 1kHz.

peagerfw, 2019-05-02:
Now i use 1Khz but i don know why burn bad.
I read all topic but i cant find answer for my problem.
what must stetting for this?

[image]

arkypita, 2019-05-02:
> > Never try with 1KHz...why???
> 
> I don't think Arkypita meant that you should'nt try it.
> I think he meant that he's never tried with 1kHz.

right

peagerfw, 2019-05-02:
arkypita you know answer to my question?

arkypita, 2019-05-02:
Does not seem to me that your laser work badly.

peagerfw, 2019-05-02:
You see all picture? :)

peagerfw, 2019-05-02:
my 9Khz laser work on 8Khz total bad...i cant setting
Now work with 1Khz but its not ideal too

arkypita, 2019-05-02:
On the first imsge (overburnd edges)
Have you set laser mode ON in grbl conf? Did you use M3 or M4?

Google for "grbl laser mode"

About jagged edge in last picture i think you have loose belt. Try tight them.
If no succes configure LaserGRBL for unidirectional engraving, use higher resolution (line/mm) with more speed.

You are the only one who can found what the best setup for your specific hardware.

peagerfw, 2019-05-02:
I Use M4 Laser mode is on.
[image]

peagerfw, 2019-05-02:
or here??? please help....
[image]

peagerfw, 2019-05-02:
$$ <
$0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=800.000
$101=800.000
$102=800.000
$110=1800.000
$111=1800.000
$112=1500.000
$120=40.000
$121=40.000
$122=40.000
$130=300.000
$131=150.000
$132=44.000

arkypita, 2019-05-02:
> or here??? please help....

There are a lot of issue talking about possible cause of image distortion. I suggest start reading them: 

https://github.com/arkypita/LaserGRBL/issues/567
https://github.com/arkypita/LaserGRBL/issues/557
https://github.com/arkypita/LaserGRBL/issues/60
https://github.com/arkypita/LaserGRBL/issues/55
https://github.com/arkypita/LaserGRBL/issues/118

and many more

peagerfw, 2019-05-06:
Thank you i read all.but not answer for my problem...
please if i have setting S30=10000
Why cant sett more as 1000 in Smax?
What must change and where?
[image]

StuartB4, 2019-05-06:
Are you sure $30 is set to 10000 or higher.
Did you click the Write button in Grbl Configuration settings after changing $30.
I just tried it and it works fine.
I set $30 to 12000 and then set S-Max to 11999 and it works.

peagerfw, 2019-05-06:
thank you ,my fault ;)

StuartB4, 2019-05-06:
No problem. :)

peagerfw, 2019-05-07:
Is possible change PWM frequency to 9Khz ?

JIOsuna, 2019-05-26:
Hi all,
Are y

JIOsuna, 2019-05-26:
Hi all,
Sorry for the last commentary. I push bottom very fast ;).
Sorry for my English. 
I have the same problem wirh my woodpecker 3.2A. I read this and i can apply this changes. If it works, i'll write here a comment.
Thanks to all for your helps.
Peagefw are you spanish or speak spanish?

JIOsuna, 2019-05-28:
Works, works!!!!! 
Thx.

peagerfw, 2019-05-29:
> Hi all,
> Sorry for the last commentary. I push bottom very fast ;).
> Sorry for my English.
> I have the same problem wirh my woodpecker 3.2A. I read this and i can apply this changes. If it works, i'll write here a comment.
> Thanks to all for your helps.
> Peagefw are you spanish or speak spanish?

I am from Slovakia

closed 2019-11-03 by arkypita

dradonja, 2020-02-02:
Hello, I'm from Serbia and I'm new to this story with a cnc laser. Buy a 50k40 machine with a 15v laser on your own. Engraving on the image and things like that quickly overcame it because the program is highly customizable for users with weaker programming behavior. So that we can help with the parameters when cutting light materials and engraving on metal surfaces. I have a black and white 3-pin connector on the board to connect the laser, and a 3-pin connector is mounted on the laser. When you turn on the white laser radio constantly without shutting down, but do not follow the commands. When you turn on the red laser for the commands but cannot cut the lake materials. I don't think we use enough laser power. Can you help me? Thanks in advance