arkypita/LaserGRBL #628: What causes the job to half size?

lloyd1941, 2019-04-27:


lloyd1941, 2019-04-27:
Gyday guys 
am new to this forum , have a EleksMaker Manna board and using LaserGRBL controller, what great and simple program. OK I set the size of the job on LG  press go and the job comes out half size ,what is wrong. is it the so called  **jumpers** in the wrong place or what, ?? stepping? trying to understand,
please explain in simple talk as I am 78, and lov this stuff!
Lloyd

arkypita, 2019-04-27:
LaserGRBL only knows measurements in millimeters, and speaks in mm when sending data to the engraver via G-Code.

A software called "grbl" run inside a chip called "atmega328" on control board.
This software is responsible to convert millimeters into motor pulses/commands, so this is the point where a conversion from mm to number of motor step will occur.

So grbl needs to know "how many steps correspond to a millimeter".
This is done by configuring grbl variables $100, $101 and $102

@StuartB4 has done a nice guide and spreadsheet to compute the correct values.
You can find it here: https://github.com/arkypita/LaserGRBL/issues/589

arkypita, 2019-04-27:
The exact number of step for mm is the result of a computation between motor step angle, gear/pulley and belt ratio, and motor driver microstepping jumpers setup.

Micro-stepping can improve resolution of your hardware, because can drive motor for half/quarter/eight or smaller angle instead of its own natural step angle. Microstepping is configured by hardware changing some jumpers on control board and will result in different values for $100 etc.

Nice calculator can be found here: https://blog.prusaprinters.org/calculator/

lloyd1941, 2019-04-27:
Thanks Arkypita
it seem that I am on the write path, but how do I adjust the settings in GRBL? 
Have a similar problem with my CNC router (I make signs) which does the same thing 50% smaller, have learnt to adjust the sizes to suit, job gets done in a way?
I also see that I am not alone in this quest?
Lloyd
Ps how do u post pic's on this site??

StuartB4, 2019-04-27:
Hi Lloyd1941.
To change the settings in GRBL you need to start LaserGRBL and click Connect.
Go to the Grbl menu top left. Click GRBL Configuration.
Look down the list for $100, $101 and $102.
Click in the Value box and edit the number to match the steps/mm that you 
want to use for your machine. 
They can be found by using the Spreadsheet in the link at the bottom
of Arkypita's first comment.
https://github.com/arkypita/LaserGRBL/issues/589

After you have entered the correct values click the Write button at the bottom of the window.
That will save your settings in GRBL.
 
[image]

StuartB4, 2019-04-27:
To add pic's here just click along the text at the bottom of the editing window when you are posting something and select your pic.
[image]

lloyd1941, 2019-04-28:
Thanks Mate,
now have your Spread sheet.  I opened the configuration file to find that $100-2-3 are already set to 80.
 so I went to the Jogging buttons and set it to 50mm. using a ruler measured the jog , it was **25mm??**
I am thinking that it is the manna board that's needs adjusting some how, jumpers,? under the drivers?
Lloyd

lloyd1941, 2019-04-28:
OK it seems that u guys have lost your voice?
In checking the new Mana board: under the Nano is a built in switch , set to off, as it comes. turn it on and we burn the Nano, to many volts. there are no jumpers under the Drivers. so whats the answer to this simple question??

StuartB4, 2019-04-28:
Did you see this picture I posted, tells you exactly what to do. 
Enter the distance you want to move. (25 mm for example) first green box, press enter.
Tell your machine to jog 25mm.
Measure the actual distance travelled.
Enter the actual distance travelled in 2nd green box.
Enter the steps/mm from your grbl configuration settings, in your case 80.
Press enter and it will give you the correct number to type in to grbl configuration settings.

[image]

lloyd1941, 2019-04-29:
Gyday Stuart
yes I have your Exl prog good job
 but the GRBL  is correct @ $100  is 80, but it stills comes out at 50%
less, their are no Jumpers on my new Mana Board, only a switch.
I( am so confused with this, and thanks to talk on email, mate

lloyde

arkypita, 2019-04-29:
If $100 is 80 and engraving still comes out at 50% of size, just set $100 to 160.

lloyd1941, 2019-04-29:
Thanks mate
 I did think in that way but wasn't shore, maybe mite damage something?
will try, and return back tomorrow nite as it is now here.
I like this way of communication , as it personal and to the point.
 I have been changing sizes to suit my jobs, it works but is not accurate?/

Lloyd
PS I live in Aussie

lloyd1941, 2019-04-30:
Thanks Arkypita
I doubled the stepping as you said, from my calc! 80 to 160, and it worked??
but what worries me is all these calculation charts, etc. didn't work for my machine, so I still ask why?? did this happen Half Size?? also I have a cnc router home built an a couple
[image]
 of 3D printers that built them.
[image]

closed 2019-04-30 by lloyd1941

arkypita, 2019-04-30:
> PS I live in Aussie

Yes, I had deduced it from the various "g'day", "Thanks Mate" & co
I was a month in Aussie some year ago. I miss its nature and the kangaroo bbq!

> but what worries me is all these calculation charts, etc. didn't work for my machine, so I still ask why?? did this happen Half Size?? also I have a cnc router home built an a couple

Maybe because you are downunder :-)

More probably because you use them the wrong way or you made mistake with your motor step angle or microstepping setup.

I have personally checked the one developed by @StuartB4 (Stepper-Calibration.xlsx) and it work: https://my.pcloud.com/publink/show?code=XZdldd7ZfaIhSLfgCauotkKlC347uYxRxFBX

This sheet simply do a proportion between requested movement and obtained movement to compute the right step/mm if actual step/mm does not give the correct result. This was the same calc I execute in my mind when i say "set to 160". It is an empiric way, but it work.

Other calculator like https://blog.prusaprinters.org/calculator/ also works. Maybe if you give a wrong input for microstepping is easy to half/double the result.

arkypita, 2019-04-30:
nice hardware