arkypita/LaserGRBL #629: Incomplete Cuts

Reynvann, 2019-04-27:
I have made several attempts to cut this shape, but many segments do not cut completely. The laser turns on as soon as travel starts, but sometimes the cutting doesn't. I read posts 192, 189, 197 and 64 but with my rudimentary understanding, I'm not sure if there is a fix for this? I can see that it was suggested having a short delay at the start of a cut segment. Was this ever implemented? If it is something I need to change in the settings file or configuration files, how can I do this? though it may appear all the uncut portions are the same length, there are man different length for .3mm up to 3mm. If any one can help resolve this I would be most appreciative. 2500mw
[image]

AlexiaPoub, 2019-05-01:
what settings did you have to cut;  I do not cut myself all the time.  it just makes engraving.

e6910, 2019-05-04:
Are you using inkscape? I had a similar problem, make sure that all paths are closed. Select on of the unfinished shapes and fill it with color. If it doesn't there's the problem. If it does fill, let's keep on looking...

AlexiaPoub, 2019-05-04:
All  Path are closed, but the laser can not burn the wood deeply and cut it off.  with the effect of a coat only to scratch ...

Στάλθηκε από το iPhone μου

arkypita, 2019-05-04:
I think that your laser have a little delay, between when he recive the power on signal and when he effectively turn on.

Maybe this could depends by some capacitors on the laser-driver control board.

You should check if you have pwm input to control laser on and off, this should act instantly

AlexiaPoub, 2019-05-04:
I can not understand what I need to Check? what is the pwm?

Στάλθηκε από το iPad μου

e6910, 2019-05-04:
What laser / controller are you using?

AlexiaPoub, 2019-05-04:
I do not know🙈 ... how can I see it?  do these photos help?

Στάλθηκε από το iPhone μου

e6910, 2019-05-04:
Please post pictures to see if we can ID your machine.

AlexiaPoub, 2019-05-04:
Help this photos?

Στάλθηκε από το iPhone μου

e6910, 2019-05-04:
No picture...

StuartB4, 2019-05-05:
[image]

Reynvann, 2019-05-05:
I'm  the original poster. Sorry for delay, I was away for work. Here is a screenshot of the original configuration.
I am primarily using jpg files with LaserGRBL. My laser module is an inexpensive 2500mw with transistor trigger, no capacitors. The parts that don't cut are always at the beginning of a cut segment and the laser is on. Even though I have my laser pinpoint focused, at the start of the segment the laser isn't pinpoint, it's larger. Then it will go back to a pinpoint and cut immediately. I have an extra pair of goggles ordered and when I get them I will try and make a video of what occurs.
[image]
tion

Reynvann, 2019-05-05:
Here are the settings. In Raster Import, PWM is checked, Unidirectional Engraving is un-checked.

[image]

Reynvann, 2019-05-05:
Also, I have used Inkscape to fill shapes, manipulate images, etc., but I have to save files to either jpg or png. That is the only way I can get anything in to LaserGRBL. Am I missing something or perhaps doing something wrong? .svg files won't open in LaserGRBL...

e6910, 2019-05-05:
IN inkscape, make sure you convert all objects to paths. on the paths menu.
then groupp the layout (ctrl-G) and align with the page. Then try and open
the svg again.

Reynvann, 2019-05-05:
Is there a certain resolution I should save images as? Initially, I think it was set to 90dpi. I changed it to 300. Would this cause any issues?

Reynvann, 2019-05-05:
> 
> IN inkscape, make sure you convert all objects to paths. on the paths menu. then groupp the layout (ctrl-G) and align with the page. Then try and open the svg again.
> […](#)

I will try this on my next project attempt. Thank you!

StuartB4, 2019-05-05:
@Reynvann.

Have you enabled Laser mode in your GRBL settings.
In the image of the settings you posted I noticed that $32 is still set to 0
Try setting it to $32=1

AlexiaPoub, 2019-05-05:
How can i change $32=1? 

Στάλθηκε από το iPad μου

AlexiaPoub, 2019-05-05:
I change $32=1 but the Problem Not resolved. 🙁 Maybe my Laser can Not cutting.

StuartB4, 2019-05-05:
Start LaserGRBL
Make sure your machine is connected by clicking the Connect button.
Click Grbl menu and select Grbl Configuration.
Scroll down to $32 and click the box with the 0 in it.
Change it to 1.
Click Write and then Close as in the picture below.
[image]

[image]

tomturican, 2019-05-06:
My problem was the treshold. I used my oszilloskop to find it out.
I build my lasedriver self. It is a CC Driver with an opamp. With 1khz PWM it can cover nearly the wohle PWM range. Sounds good, but the problem is the laser itself. He need a minimum energy (or with pwm a minimum duty cycle) to work. My 50mw UV laser begins at 4% dyty cylce and my 2,5W laser needs a minimum from 8% duty cycle. The second problem was, that the Minimum spindle speed setting simply not works with my unchanged 1.1f grbl. (now i changed the config.h and enabled the setting with a value of 20)
OK... its a long intro.... ;-). The problem occurs by me, if i used a too high working speed. GRBL will change the laserpower and adapt it to the speed. But with my slow acceleration settings and the high detailed structures my machine never reached this speed and GRBL changed the laserpower below the treshold and the laser switched off. I changed the config.h file and enabled the min-spindle setting with auf vale of 20 (enough to sure switch the laser on)....

Reynvann, 2019-05-07:
> 
> 
> @Reynvann
> 
> Start LaserGRBL
> Make sure your machine is connected by clicking the Connect button.
> Click Grbl menu and select Grbl Configuration.
> Scroll down to $32 and click the box with the 0 in it.
> Change it to 1.
> Click Write and then Close as in the picture below.
> [image]
> 
> [image]

I have previously set laser to 1 with same results. Yes, this is what one could consider a cheap Chinese laser. I have purchased a new much larger machine with a 7 watt laser. My original machine is a 4030 with 2.5 watt, but as I said before, I think the laser module is neither 2.5 watt nor good. I think it might be junk!

closed 2019-11-03 by arkypita