arkypita/LaserGRBL #631: CNC Shiled V3.0 and problem with motor speed

Pietrogramma, 2019-04-29:
Hi everyone, I built an MPCNC to which, I connected a CNC Shiled with DRV8825 driver.
Following some tutorials, I set the microsteps of the card to 1/32 (all three PINs with jump) and I entered the optimal values ​​for each single axis, by command $ 100 (for x axis); $ 101 (for Y axis) and $ 102 (for Z axis). 
Now the displacements for each single axis are perfect to the millimeter (if I give the command to move the Y axis of 100 the carriage makes me a precise 10 cm (measurement made with the digital caliper).
The only problem is that the motors run really slowly, but very very slowly.
How can I increase the speed of the motors? if I try to change the Gcode F command, the speed remains unchanged. 
Hello and thanks

arkypita, 2019-04-29:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#110-111-and-112--xyz-max-rate-mmmin

Pietrogramma, 2019-04-30:
Thanks arkypita, now it work perfectly.
Regards

closed 2019-05-06 by arkypita