arkypita/LaserGRBL #634: Low power on laser 5500mw

spriggam, 2019-05-01:
Hi , i am a beginner
I ve a laser engraver buy on ebay , it 's a 5500mw laser with arduino Nano 
The grbl is 0,9 
my problem is that when I run a burner the laser is very weak even if I vary the values ​​S to 255 . i can´t cut paper
Could you help me  ?

StuartB4, 2019-05-01:
Is it a new machine or used?
Have you tried it with different software?

spriggam, 2019-05-02:
it s a new laser , i try Benbox before but it doesn t work at all

StuartB4, 2019-05-02:
You may have a faulty laser or laser driver.

spriggam, 2019-05-02:
i don t think it s a problem of laser i think it s a problem of command 
but i don t know how to do
do you know another software i could try ?

StuartB4, 2019-05-02:
You could try T2 Laser. They have a free trial version so you don't have to pay to try it.
http://t2graphics.weebly.com
Also look on youtube for how to use it.
https://www.youtube.com/watch?v=lyxqht64JNQ

osmosis14, 2019-05-03:
Its not the laser, change the power from 255 to something more like 900 or 1000

spriggam, 2019-05-04:
i try to change the power to 900 , but it doesn t work
and i try T2 software and it s work 
thanks a lot

Ersukon, 2019-05-04:
Take a look

[image]

spriggam, 2019-05-04:
Thank you , you are right , it s work perfectly

closed 2019-05-06 by arkypita