arkypita/LaserGRBL #637: jogging not working

majstoralc9, 2019-05-05:
Hello
I upgrated laser grbl program but now jogging not working.
I try everything.
Can anybody help me?
Best regards
[image]

arkypita, 2019-05-06:
Where did you download?
Did you compile from source? If yes, please update to latest trunk version

majstoralc9, 2019-05-06:
I download program from github.com. Before upgrade jogging works well, but after upgrade it's like in atached picture.

arkypita, 2019-05-07:
For me, seem to work

[image]

arkypita, 2019-05-07:
However I am going to release a new version to restore prev version of jog control

majstoralc9, 2019-05-07:
Thanks for reply. In my case jogging working only with mouse conntrol on screen and not working with keyboard conntrol.
When i press for example key 6 - for x movement give me program code $9=G91x0y0
In your case is code diferent.
Sorry for my bad english.
Best regards

arkypita, 2019-05-07:
Oh, ok. Yes, this was already reported.
Just wait for new version, I'm fixing it.

closed 2019-05-07 by arkypita

arkypita, 2019-05-07:
Whould you like to try? https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.11

majstoralc9, 2019-05-10:
Its working. Thanks