arkypita/LaserGRBL #639: No linear movement when gcode changes from M3/M4 to M5 (LaserMode)

Naisk2k, 2019-05-06:
Hi,
i've got nearly the same problem as #318 . But LaserMode is active and only when i use dithering (without PWM). It makes the process really slow (5h instead of 1h). It works fine and steady when i change the Gcode. 
M3 S0 for M5 
M3 S255 for M3 and
G1 for all G0

Best regards.

arkypita, 2019-05-06:
verify max speed and acceleration in GRBL configuration
Is not a LaserGRBL issue, so if issue persist discuss in GRBL github tracker, not here

Naisk2k, 2019-05-06:
Thank you for your fast answer.
Maybe it is not a real issue rather an improvement. But (from GRBL wiki - LaserMode) :

Ensure smooth motions throughout by turning the laser on and off without an M3 M4 M5 spindle state command. 
(https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode - CAM Developer Implementation Notes)

And this what LaserGRBL does when dithering Images, switching between M3 and M5 plus using G0. As a result, the maximum speed is not reached even if the acceleration is set higher. The gcode does not use lasermode, so to speak.

Sorry, i'am not a pro and don't understand or "find" everything. I also tried it in the GRBL github tracker but without a result.

closed 2019-11-03 by arkypita